gpiozero/gpiozero #957: intermittent bug in pigpio factory, sl.s.send AttributeError: 'NoneType' object has no attribute 'send'

ddan39, 2021-04-15:
Operating system: Raspbian 10.9
Python version: 3.7
Pi model: Pi 4 Model B
GPIO Zero version: 1.6.2
Pin factory used: pigpio

simply creating a Button() , then sleep for 0.01 seconds and exit the script, and once in a while it will crash with this traceback:

Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/gpiozero-1.6.2-py3.7.egg/gpiozero/mixins.py", line 546, in held
    parent._held_from = parent.pin_factory.ticks()
  File "/home/pi/.local/lib/python3.7/site-packages/gpiozero-1.6.2-py3.7.egg/gpiozero/pins/pigpio.py", line 154, in ticks
    return self._connection.get_current_tick()
  File "/usr/lib/python3/dist-packages/pigpio.py", line 2030, in get_current_tick
    return _pigpio_command(self.sl, _PI_CMD_TICK, 0, 0)
  File "/usr/lib/python3/dist-packages/pigpio.py", line 993, in _pigpio_command
    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
AttributeError: 'NoneType' object has no attribute 'send'

systemdarena, 2021-04-15:
looking at this again... i wonder if the bug is actually in pigpio library? though i never saw this when using 1.5.1

joan2937, 2021-04-16:
It's certainly raised by pigpio.  It's the error you would get if the connection to the pigpio daemon (in the pigpio Python module) has been closed.  Given that you are exiting the script it sounds like a shutdown race condition.  Within the pigpio Python module the connection to the daemon thread is closed before the main thread.  Perhaps the pigpio Python module has received a close command shortly after followed by the get_current_tick() command.