gpiozero/gpiozero #958: Multi sensor line follower

GeekyTim, 2021-05-08:
Device name: Line Follower
Protocol: Digital
URL of example product:
Further information e.g. data sheet:

Can the current line-follower class be extended to handle 2/3/4/5 sensor line followers?

lurch, 2021-05-09:
Hi Tim :wave: 

It's easy to build a [`CompositeDevice`](https://gpiozero.readthedocs.io/en/stable/api_boards.html#compositedevice) of multiple [`LineSensor`s](https://gpiozero.readthedocs.io/en/stable/api_input.html#linesensor-trct5000), and (IIRC) this will give you a `.value` of a tuple of floating-point values (one for each LineSensor).
Is that sufficient? What would you expect the "API" for a 4-sensor line follower to look like?

GeekyTim, 2021-05-09:
TBH I had not seen CompositeDevice before, and the API I think would be pretty much what that would output.