gpiozero/gpiozero #963: mock pin example in documentation 1.6 does not work

beetlegigg, 2021-05-21:
Operating system: Latest rpi os as at report date
Python version:  3.7
Pi model:  Pi 4 Model B
GPIO Zero version:  1.5.1
Pin factory used: e.g. MockFactory

Please give us as much information about your issue as possible. Write your
code inside code blocks like so:

see the gpiozero documentation v1.62 on mock pins and the example code. The example given uses the REPL but I have put the same code into a file as below and included a small bit of text in the printout of the led.value. All the comments in the code are from the example on mock pins found in the docs at section 23.2

```python
from gpiozero import Device
from gpiozero.pins.mock import MockFactory
Device.pin_factory = MockFactory()

from gpiozero import LED, Button

# Construct a couple of devices attached to mock pins 16 and 17, and link the devices
led = LED(17)
btn = Button(16)
led.source = btn

# Initailly the button isn't "pressed" so the LED should be off
print('led.value, expect 0, result = ' , led.value)

# Drive the pin low (this is what would happen electrically when the button is pressed)
btn.pin.drive_low()
# The LED is now on
print('led.value, expect 1, result = ' , led.value)

btn.pin.drive_high()
# The button is now "released", so the LED should be off again
print('led.value, expect 0, result = ' , led.value)

```

When the program is executed I get and the LED does not show as on when the button is 'mock' pushed
led.value, expect 0, result = 0
led.value, expect 1, result = 0
led.value, expect 0, result = 0

bennuttall, 2021-05-21:
This is just because the `led.source = btn` has a delay of 0.1 seconds (by default). Here the LED is polling the button for its value every 0.1 seconds and setting itself to that value. But as each line of code is executed, this background thread hasn't updated it yet.

In the example, it's assumed time passes between each line being executed. Adding a sleep works:

```python
from gpiozero import Device
from gpiozero.pins.mock import MockFactory
from time import sleep
Device.pin_factory = MockFactory()

from gpiozero import LED, Button

# Construct a couple of devices attached to mock pins 16 and 17, and link the devices
led = LED(17)
btn = Button(16)
led.source = btn

# Initailly the button isn't "pressed" so the LED should be off
print('led.value, expect 0, result = ' , led.value)

# Drive the pin low (this is what would happen electrically when the button is pressed)
btn.pin.drive_low()
sleep(0.1)
# The LED is now on
print('led.value, expect 1, result = ' , led.value)

btn.pin.drive_high()
sleep(0.1)
# The button is now "released", so the LED should be off again
print('led.value, expect 0, result = ' , led.value)
```

Output:

```
led.value, expect 0, result =  0
led.value, expect 1, result =  1
led.value, expect 0, result =  0
```

I tried setting `led.source_delay` to 0 but that isn't fool proof because it's still a background thread that has to loop. For me it actually resulted in 0, 0, 1 so that's not a solution.

Alternatively, using event-based device connections seems to work too:

```python
from gpiozero import Device
from gpiozero.pins.mock import MockFactory
Device.pin_factory = MockFactory()

from gpiozero import LED, Button

# Construct a couple of devices attached to mock pins 16 and 17, and link the devices
led = LED(17)
btn = Button(16)

btn.when_pressed = led.on
btn.when_released = led.off

# Initailly the button isn't "pressed" so the LED should be off
print('led.value, expect 0, result = ' , led.value)

# Drive the pin low (this is what would happen electrically when the button is pressed)
btn.pin.drive_low()
# The LED is now on
print('led.value, expect 1, result = ' , led.value)

btn.pin.drive_high()
# The button is now "released", so the LED should be off again
print('led.value, expect 0, result = ' , led.value)
```

Output:

```
led.value, expect 0, result =  0
led.value, expect 1, result =  1
led.value, expect 0, result =  0
```

TODO: add a note to the docs about this

lurch, 2021-05-21:
> The example given uses the REPL

Obviously when using the REPL we're assuming that the user will take more than 0.1 seconds to type in the next line of code :rofl:

beetlegigg, 2021-05-21:
Thanks bennuttall that explains it, and thanks for initiating a document update and giving example working code.  Much appreciated.

lurch, yes obviously using the REPL would take more time to enter the command, but perhaps even more obvious it that using the mock pins would be used within a script to test the code, and the relevant section of the document did not indicate the nature of setting a button pin high or low would need a delay.  Whilst the relevant info may be found somewhere in the documentation,  a new user of GPIO zero, like me, would naturally only be focusing on a relevant bit of the document when endeavouring to understand a particular aspect of the library.   I would further suggest that giving examples would be much better given as blocks of code that can be cut and pasted rather than a bunch REPL statements.

lurch, 2021-05-21:
@beetlegigg Please don't take offence; I wasn't laughing at you, I was laughing at the nature of our instructions :slightly_smiling_face: 
I totally agree that your concerns are legitimate.