gpiozero/gpiozero #966: List/table of features by pin factory

bennuttall, 2021-05-27:
It would be nice if we had a list or table showing what the features of each pin factory are, e.,g. hardware PWM - perhaps on the pins page of the docs: https://gpiozero.readthedocs.io/en/stable/api_pins.html

lurch, 2021-05-27:
Good idea. Such a table could also list which models of Pi each pin factory works with.

bennuttall, 2021-05-27:
Probably no need as we've removed RPIO (as of the upcoming 2.0 release): https://gpiozero.readthedocs.io/en/master/compat.html#rpio-pin-factory-removed

lurch, 2021-05-27:
Oooh, I'll have to have a look at all the new shininess at some point... :sunglasses:

lurch, 2021-05-27:
ping @waveform80 
Nice to see that Motors and Robots have been cleaned up :+1: 
Might also be worth renaming https://gpiozero.readthedocs.io/en/master/api_spi.html#analoginputdevice to `AnalogInputSPIDevice` (in preparation for a possible `AnalogInputI2CDevice` being added in future)? And I think I once suggested that it might make sense for e.g. https://gpiozero.readthedocs.io/en/master/api_spi.html#mcp3004 to be a single device with 4 sub-devices, rather than explicitly passing in a `channel` number (and duplicate `spi_args`) to every `MCP3004` you create?
It'd probably be useful if the layout-diagrams shown at https://gpiozero.readthedocs.io/en/master/api_boards.html#ledchardisplay were also shown at https://gpiozero.readthedocs.io/en/master/api_fonts.html ?
Should https://gpiozero.readthedocs.io/en/master/api_boards.html#traffichat and https://gpiozero.readthedocs.io/en/master/api_boards.html#fishdish extend https://gpiozero.readthedocs.io/en/master/api_boards.html#trafficlightsbuzzer instead of `CompositeDevice`? (and what's the distinction between `LEDCollection` being a "Base Class" but `TrafficLights` being a "Regular Class"?)
Does https://gpiozero.readthedocs.io/en/master/api_boards.html#phaseenablerobot need to be removed? (and PhaseEnableRobot is still in https://gpiozero.readthedocs.io/en/master/api_boards.html#base-classes too)