gpiozero/gpiozero #972: motion_detected fails with exception - "RuntimeError: deque mutated during iteration"

dbrandesky, 2021-06-14:
I have a fairly "simple" loop on a Raspberry Pi Zero W that runs two applications: the first basically generates random "arrivals" of a simulated train and activates GPIO pins when those events occur; the second is a function that monitors the state of an external motion detector, and passes that state into a function which will return True if motion is detected.

The exception is as follows:
`Traceback (most recent call last):
  File "TrainGenerator.py", line 107, in <module>
    detector_state = garage_motion(mqttc, topic_state, motion_sensor
)
  File "TrainGenerator.py", line 54, in garage_motion
    if sensor.motion_detected:
  File "/home/pi/TrainGenerator/venv/lib/python3.7/site-packages/gpi
ozero/input_devices.py", line 350, in is_active
    return self.value > self.threshold   
  File "/home/pi/TrainGenerator/venv/lib/python3.7/site-packages/gpi
ozero/input_devices.py", line 604, in value
    return super(MotionSensor, self).value
  File "/home/pi/TrainGenerator/venv/lib/python3.7/site-packages/gpi
ozero/input_devices.py", line 325, in value
    return self._queue.value
  File "/home/pi/TrainGenerator/venv/lib/python3.7/site-packages/gpi
ozero/mixins.py", line 587, in value
    return self.average(self.queue)
  File "/usr/lib/python3.7/statistics.py", line 311, in mean
    T, total, count = _sum(data)
  File "/usr/lib/python3.7/statistics.py", line 147, in _sum
    for n,d in map(_exact_ratio, values):
RuntimeError: deque mutated during iteration
`

The offending function is this:
`def garage_motion(mqtt_client, topic, sensor):
    if sensor.motion_detected:
        mqtt_publish(mqtt_client, topic, 'ON')
        return True
    else:
        mqtt_publish(mqtt_client, topic, 'OFF')
        return False
`

This function is run every 0.1 seconds in the main loop.

Full code can be found here: [](https://pastebin.com/6zXvvWnX)

I'm totally at a loss here, any tips would be appreciated.

bennuttall, 2021-06-14:
> EDIT: I have no idea why the line breaks all got lost in my code snippets.

Use three backticks to fence your code, not one (I've edited it)

bennuttall, 2021-06-14:
What version of gpiozero is this?

dbrandesky, 2021-06-14:
> What version of gpiozero is this?

Version 1.6.2 running on Python 3.7.3.

bennuttall, 2021-06-14:
It looks like you're polling too fast for it to handle. You could try waiting a little longer.

Alternatively you could try another method of triggering changes from the sensor:

```python
sensor.when_motion = publish_on
sensor.when_no_motion = publish_off
```

or

```python
while True:
    sensor.wait_for_motion()
    mqtt_publish(mqtt_client, topic, 'ON')
    sensor.wait_for_no_motion()
    mqtt_publish(mqtt_client, topic, 'OFF')
```

dbrandesky, 2021-06-14:
> It looks like you're polling too fast for it to handle. You could try waiting a little longer.
>
> Alternatively you could try another method of triggering changes from the sensor:
>
> ```python [...]

I definitely don't mind a longer wait. I currently wait for 0.1 seconds, and the code ran for about two days before the exception in that test.

I wasn't really clear on the wait_for_motion() method; it seemed like it was a blocking call; i.e. my loop would pause until motion occurred.

bennuttall, 2021-06-14:
> I wasn't really clear on the wait_for_motion() method; it seemed like it was a blocking call; i.e. my loop would pause until motion occurred.

Yes, it's quite useful if that's compatible with your program flow. There are 4 different ways of reacting to device changes: polling, blocking, event-driven and source/values. See the [FAQ](https://gpiozero.readthedocs.io/en/stable/faq.html#what-s-the-difference-between-when-pressed-is-pressed-and-wait-for-press) which covers this topic for buttons, and it's much the same for other devices.

closed 2021-06-14 by bennuttall