gpiozero/gpiozero #974: RotaryEncoder comments

lurch, 2021-06-15:
Following on from https://github.com/gpiozero/gpiozero/issues/971 I've just read the [RotaryEncoder](https://github.com/gpiozero/gpiozero/blob/master/gpiozero/input_devices.py#L982) code for the first time. I thought you might like some feedback... ping @waveform80 

The `threshold_steps` feels a bit weird - I can't quite tell how it was intended to be used, but I wonder if it might have made sense for the device to be "active" when it's below the min_value (i.e. turned sufficiently far counter-clockwise) **or** above the max_value (i.e. turned sufficiently far clockwise), with values in the "middle" being "inactive"? Whereas currently (AFAICT) the logic is inverse to this where the "middle" is active, and values outside of this (i.e. turned further to the left or right) then make it "inactive"??

The docstring for `max_steps` should probably mention its default value of 16?

Also (again, AFAICT from only having read the code rather than actually experimenting with it), the RotaryEncoder treats turning to the left as "negative values" and turning to the right as "positive values". While this is a perfectly understandable API, I wonder if it'd be worth adding a subclass or something that worked like a "volume control knob"? I.e. turning to the left only counts down to 0, but won't decrease any further if turned CCW beyond that, and turning to the right will increase to e.g. 100 but won't increase any further if rotated CW beyond that? :shrug: 

And finally, it's not mentioned in the API docs but I guess if the user wires up their RotaryEncoder and turning it clockwise actually triggers the CCW code in gpiozero (and vice-versa), they can simply fix that by swapping the order of the two GPIO pins passed to the constructor?

And finally finally, scaled_half seems to be missing from https://gpiozero.readthedocs.io/en/latest/api_tools.html

waveform80, 2021-06-16:
> Following on from #971 I've just read the [RotaryEncoder](https://github.com/gpiozero/gpiozero/blob/master/gpiozero/input_devices.py#L982) code for the first time. I thought you might like some feedback... ping @waveform80

Always welcome!

> The `threshold_steps` feels a bit weird - I can't quite tell how it was intended to be used, but I wonder if it might have made sense for the device to be "active" when it's below the min_value [...]

Yup, it's definitely "weird". Frankly the only reason it exists is because all devices have to have a boolean `active` property (as the base class defines this), and I couldn't think of a better means of defining this than "a range of `steps` between which `active` is `True`". At the moment it's exactly as you describe. For instance, when `max_steps` is 5 (which permits `steps` to vary between -5 and 5) and `threshold_steps` is `(1, 5)` then:

* Encoder starts off "inactive" as `steps == 0`
* We rotate it clockwise one notch; now `steps == 1` and the encoder is "active"
* We rotate it two notches anti-clockwise; now `steps == -1` and the encoder is "inactive" again

One enhancement I'd considered for the future was permitting `threshold_steps` to be an inverted range, e.g. `(5, 1)` which would indicate that values *outside* the specified range would be `True`, but I decided the initial implementation should be as simple as possible (given this extension would be backwards compatible, I saw no particular harm in leaving it out for now).

> The docstring for `max_steps` should probably mention its default value of 16?

Yes, it probably should -- that'll be included in the [generated sphinx docs](https://gpiozero.readthedocs.io/en/stable/api_input.html#rotaryencoder) anyway (iun the class prototype) but it never hurts to make it explicit for those using `help()` at the REPL.

> Also (again, AFAICT from only having read the code rather than actually experimenting with it), the RotaryEncoder treats turning to the left as "negative values" and turning to the right as [...]

Heh, this was another thing that I'd considered when designing it. The tentative proposal for the future was to extend the `max_steps` definition to permit a similar `(min, max)` range to `threshold_steps` (i.e. a single integer, as specified currently, would be interpreted as `(-n, n)` in this system), hence users could specify (0, 10) to achieve the "volume control" scenario.

The only wrinkle with that was how to deal with the limits of the `value` conversion. Should there be a special case that when the "min" of `max_steps` is 0, the minimum of `value` should likewise be 0 instead of -1? I didn't particularly like the special-casing but it also made a sort of intuitive sense. Or would it be preferable to have a boolean "minimum is zero" flag that simply changed `max_steps` to mean a range of `(0, max_steps)` instead of `(-max_steps, max_steps)`? It's less flexible (no ability to specify an arbitrary `(n, m)` range) but probably fits most real-world scenarios better.

Ultimately it was another case of "keep the initial implementation simple", but I'd be interested to hear thoughts on the different options!

> And finally, it's not mentioned in the API docs but I guess if the user wires up their RotaryEncoder and turning it clockwise actually triggers the CCW code in gpiozero (and vice-versa), they can simply fix that by swapping the order of the two GPIO pins passed to the constructor?

Yes, and that should indeed be mentioned in the docs!

> And finally finally, scaled_half seems to be missing from https://gpiozero.readthedocs.io/en/latest/api_tools.html

It is? It is! Damn...

lurch, 2021-06-16:
> > Following on from #971 I've just read the RotaryEncoder code for the first time. I thought you might like some feedback... ping @waveform80

> Always welcome!

I also left you some comments somewhere about SPI...  ....rummages.... ah, [here](https://github.com/gpiozero/gpiozero/issues/966#issuecomment-849709605) - dunno if you saw those?

>  Yup, it's definitely "weird".
One enhancement I'd considered for the future was permitting threshold_steps to be an inverted range, e.g. (5, 1) which would indicate that values outside the specified range would be True

Ah, that's a nice idea :+1:  Although a non-intuitive API :wink:  Perhaps a boolean `invert_threshold` kwarg would be better?

> Yes, it probably should -- that'll be included in the generated sphinx docs anyway (iun the class prototype) 

Ahh! I'd missed that it was mentioned in the `classgpiozero.RotaryEncoder(a, b, *, bounce_time=None, max_steps=16, threshold_steps=(0, 0), wrap=False, pin_factory=None)` line as I was only looking at the "Parameters:" section of the sphinx docs.

> Heh, this was another thing that I'd considered when designing it.

Great minds... :slightly_smiling_face: 

> Should there be a special case that when the "min" of max_steps is 0, the minimum of value should likewise be 0 instead of -1? I didn't particularly like the special-casing but it also made a sort of intuitive sense.

IMHO yes it should. The user can then just multiply the `.value` by whatever range they want to scale it to (if necessary). (Or use source-tools, of course)

> Or would it be preferable to have a boolean "minimum is zero" flag that simply changed max_steps to mean a range of (0, max_steps) instead of (-max_steps, max_steps)?

Yeah, that would probably work well :+1:

mikiair, 2021-07-14:
Hi!
I also had a first glance on the RotaryEncoder source a few days ago and I might add another comment...
As far as I understood the RotaryEncoder inits two internal InputDevice references which are hard-coded set to _internal_ pull-ups:
```
...
super(RotaryEncoder, self).__init__(
            a=InputDevice(a, pull_up=True, pin_factory=pin_factory),
            b=InputDevice(b, pull_up=True, pin_factory=pin_factory),
            _order=('a', 'b'), pin_factory=pin_factory)
...
```
From my search, it looks like there are readily wired **rotary encoder modules** available which are in turn hard-wired with _external_ pull-ups (rather not pull-downs?!). If connected to the GPIOs this would result in two pairs of parallel resistors which might bring you on the safe side, but in turn reduces total resistance.
Would it be possible to have this choice (external vs. internal) via an optional argument like for the Button class?
Or does this have further implications on the rotation logic (like the above mentioned pin exchange)?

waveform80, 2021-07-14:
Lots of good ideas here for RotaryEncoder in 2.0. I'll have a play with some of these and see how things "feel". Targetting for 2.0.

> As far as I understood the RotaryEncoder inits two internal InputDevice references which are hard-coded set to _internal_ pull-ups:

Yup, that's correct.

> From my search, it looks like there are readily wired **rotary encoder modules** available which are in turn hard-wired with _external_ pull-ups (rather not pull-downs?!). If connected to the GPIOs [...]

Yes, I can pull out the `pull_up` argument into the `RotaryEncoder` initializer. It would default to `True` (as it is currently for backwards compatibility) but could be set to `False` (for pull down, for any modules that happen to be wired that way), or `None` for "I've got my own pulls".

In the latter case, the worst case is that the rotation logic is backwards. We could either add `active_high` (as in `Button`) to permit reversal of the logic, or just add the suggestion from lurch earlier that simply reversing the A&B connections reverses the operation of the encoder anyway (I *think* that would work in this case as everything's done on edges rather than levels, but I'd probably have to sit down and think a bit harder about it to convince myself).