gpiozero/gpiozero #975: motion_detected fails with exception - "RuntimeError: deque mutated during iteration"

bennuttall, 2021-06-16:
*Originally reported in #973 by @dbrandesky and moved to discussions, but turned out to be a bug*

I have a fairly "simple" loop on a Raspberry Pi Zero W that runs two applications: the first basically generates random "arrivals" of a simulated train and activates GPIO pins when those events occur; the second is a function that monitors the state of an external motion detector, and passes that state into a function which will return True if motion is detected.

The exception is as follows:

```python
Traceback (most recent call last):
  File "TrainGenerator.py", line 107, in <module>
    detector_state = garage_motion(mqttc, topic_state, motion_sensor
)
  File "TrainGenerator.py", line 54, in garage_motion
    if sensor.motion_detected:
  File "/home/pi/TrainGenerator/venv/lib/python3.7/site-packages/gpi
ozero/input_devices.py", line 350, in is_active
    return self.value > self.threshold   
  File "/home/pi/TrainGenerator/venv/lib/python3.7/site-packages/gpi
ozero/input_devices.py", line 604, in value
    return super(MotionSensor, self).value
  File "/home/pi/TrainGenerator/venv/lib/python3.7/site-packages/gpi
ozero/input_devices.py", line 325, in value
    return self._queue.value
  File "/home/pi/TrainGenerator/venv/lib/python3.7/site-packages/gpi
ozero/mixins.py", line 587, in value
    return self.average(self.queue)
  File "/usr/lib/python3.7/statistics.py", line 311, in mean
    T, total, count = _sum(data)
  File "/usr/lib/python3.7/statistics.py", line 147, in _sum
    for n,d in map(_exact_ratio, values):
RuntimeError: deque mutated during iteration
```

The offending function is this:

```python
def garage_motion(mqtt_client, topic, sensor):
    if sensor.motion_detected:
        mqtt_publish(mqtt_client, topic, 'ON')
        return True
    else:
        mqtt_publish(mqtt_client, topic, 'OFF')
        return False
```

This function is run every 0.1 seconds in the main loop.

Full code can be found [here](https://pastebin.com/6zXvvWnX).

danneboom, 2021-07-22:
same here with a MotionSensor and polling `pir.is_active` . Any workaround?

lurch, 2021-07-22:
According to the comment [here](https://github.com/gpiozero/gpiozero/discussions/973#discussioncomment-875282), creating the `MotionSensor` with `average=statistics.median` _might_ be a workaround? (Although I've not tested)

tamberg, 2023-09-13:
Same error message from [different code](https://github.com/tamberg/pi-lora/blob/main/rn2483-pir.py) calling .motion_detected of a MotionSensor in a fast loop:

```
Traceback (most recent call last):
   File "/home/pi/rn2483-pir.py", line 66, in <module>
     if pir.motion_detected:
   File "/usr/lib/python3/dist-packages/gpiozero/input_de>
     return self.value > self.threshold
   File "/usr/lib/python3/dist-packages/gpiozero/input_de>
     return super(MotionSensor, self).value
   File "/usr/lib/python3/dist-packages/gpiozero/input_de>
     return self._queue.value
   File "/usr/lib/python3/dist-packages/gpiozero/mixins.p>
     return self.average(self.queue)
   File "/usr/lib/python3.9/statistics.py", line 316, in >
     T, total, count = _sum(data)
   File "/usr/lib/python3.9/statistics.py", line 166, in >
     for n, d in map(_exact_ratio, values):
 RuntimeError: deque mutated during iteration
```
https://github.com/gpiozero/gpiozero/blob/master/gpiozero/input_devices.py#L519
```
class MotionSensor(SmoothedInputDevice)
```

https://github.com/gpiozero/gpiozero/blob/master/gpiozero/input_devices.py#L176-L185
```
This class extends :class:`InputDevice` with a queue which is filled by a
    background thread which continually polls the state of the underlying
    device.
 ```

and

 ```
    The average (a configurable function) of the values in the queue is
    compared to a threshold which is used to determine the state of the
    :attr:`is_active` property.
```

Is there a chance that the background thread adds values to the queue while the same queue is iterated to compute the average?