gpiozero/gpiozero #976: Issue with installing gpiozero on my Ubuntu laptop

Lwk7426, 2021-06-23:
Hi, I am currently quite new to the Ubuntu OS and i am having some trouble with installing the gpiozero package onto my laptop. I tried running "sudo apt update" follow by "sudo apt install python3-gpiozero" and "sudo apt install python-gpiozero" command to install but i kept getting the results as shown below 

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package python3-gpiozero

May i know what are the steps to take to properly install the package onto my Ubuntu Laptop? Thank you very much.

lurch, 2021-06-23:
https://gpiozero.readthedocs.io/en/stable/installing.html#pc-mac

Lwk7426, 2021-06-23:
Hi lurch, thank you for providing the link. I've actually looked into this link before but I am still unable to locate the package after using the steps to install. I am not sure what i am missing out.

lurch, 2021-06-23:
Did you follow the links through to https://gpiozero.readthedocs.io/en/stable/remote_gpio.html#linux ?

bennuttall, 2021-06-24:
What version of Ubuntu?

franva, 2021-10-04:
> What version of Ubuntu?

I use Ubuntu 21.10 on RPi 4 with 4GB RAM.

To see more details: https://github.com/gpiozero/gpiozero/issues/1000