gpiozero/gpiozero #977: Option to use pigpio C library bindings instead of pigpiod

cinderblock, 2021-06-26:
At the lowest level, pigpio is a C library that talks directly to memory mapped hardware for super fast and reliable GPIO.

This library (gpiozero) seems to interface with the pigpio daemon (`pigpiod`) over a socket (over the network stack!) which is itself using the low level C library.

Have you considered writing python bindings that skip `pigpiod` and talk directly to the memory with the `pigpio` C library?

For reference, the [Node.js pigpio library](https://github.com/fivdi/pigpio) uses the C library and skips any `pigpiod`.

Of note, this does add a minor complication that you can only have one program using the `pigpio` low level C interface at a time because having multiple threads access the memory mapped GPIO hardware is a recipe for disaster (fortunately the low level API has guards for this).

bennuttall, 2021-06-27:
If you want you can write your own pin factory that interacts with the GPIO in that way. If you find it works well you can send us a PR and people can choose to use it as they need to.

See pin factory implementations here: https://github.com/gpiozero/gpiozero/tree/master/gpiozero/pins

lurch, 2021-06-28:
`NativePin` also uses memory-mapped registers, but it's probably much slower than the C equivalent :wink: 

> Of note, this does add a minor complication

I guess it would also add a requirement for native-compiled C code to GpioZero, whereas currently it's only using pure Python.

cinderblock, 2021-06-28:
@lurch that is a good point. I'm coming from the Node world where compiling and linking to C++ was a non-issue because Npm and `node_gyp` build the necessary binaries on all systems without issue. I expect there is an equivalent in the Python world but I'm not sure. Any thoughts on this would be appreciated.

lurch, 2021-06-28:
Beyond my level of knowledge, but I'm sure it's something that @waveform80 knows all about. Alternatively, ask google :wink: (which finds e.g. https://docs.python.org/3/extending/building.html but I'm sure there are also more detailed examples).