gpiozero/gpiozero #982: RuntimeError: Not running on a RPi!

pigscoven, 2021-07-18:
Operating system: e.g. Raspbian Buster
Python version: e.g. 3.7
Pi model: e.g. Pi 4 Model B
GPIO Zero version: e.g. 1.5.1
Pin factory used: e.g. RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

Please give us as much information about your issue as possible. Write your
code inside code blocks like so:
I am learning to code and went through your handy guide. Most of the recipes I tried worked awesome. However, I tried this simple code for my motion sensor and it gives me an RuntimeError: Not running on a RPi! I have it set up exactly how the directions show using one for pin 2, one for 5V, and one for ground. I have tried 5 motion sensors thinking it was something wrong with the sensors. I am not sure what the message is trying to say to me. I have researched it and I have done all that I found from researching. I am hoping someone can help me. I figured sticking to codes that work before getting into my own first would be best for learning. This one is frustrating me. I am running it with python 3. Thank you.

```python
from gpiozero import MotionSensor

pir = MotionSensor(2)

pir.wait_for_motion()
print("Motion Detected!")
print("code here")
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 101, in pin
    pin = self.pins[n]
KeyError: 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "motionStest.py", line 3, in <module>
    pir = MotionSensor(2)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 108, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 593, in __init__
    sample_rate, partial=partial, pin_factory=pin_factory, average=mean)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 262, in __init__
    pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 218, in __init__
    super(EventsMixin, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 83, in __init__
    super(InputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 103, in pin
    pin = self.pin_class(self, n)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 111, in __init__
    GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Not running on a RPi!

bennuttall, 2021-07-18:
```
Operating system: e.g. Raspbian Buster
Python version: e.g. 3.7
Pi model: e.g. Pi 4 Model B
GPIO Zero version: e.g. 1.5.1
Pin factory used: e.g. RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details
```

Can you fill in this please?

```
Most of the recipes I tried worked awesome. However, I tried this simple code for my motion sensor and it gives me an RuntimeError: Not running on a RPi! 
```

It seems unlikely that *some* of the recipes work and others don't. The error you're getting is from RPi.GPIO so this is fundamental to all functionality.

pigscoven, 2021-07-18:
Thank you bennuttall for responding. 
How do I fix the problem? Could It be the motion sensors are all broken? Is it saying that it can not find something connected to the pin to run it? 
Thank you.

bcroston, 2021-07-18:
Try:
$ pip3 install --pre RPi.GPIO

bennuttall, 2021-07-18:
@pigscoven I need this information from you:

```
Operating system: ???
Python version: ???
Pi model: ???
GPIO Zero version: ???
```

pigscoven, 2021-07-19:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspberry Pi 4 Model B Rev 1.2

from pkg_resources import require
>>> require('gpiozero')[0].version
'1.6.2'

from gpiozero import Device
>>> Device._default_pin_factory()
<gpiozero.pins.rpigpio.RPiGPIOFactory object at 0xb5179c50>

I think I found the information you needed to help.

bennuttall, 2021-07-19:
There's no reason why this would happen suddenly. I would try again after a reboot and if it continues, try again with a fresh OS image.

pigscoven, 2021-07-19:
@bennuttall 
I did not do what @bcroston suggested. Should I do that first and try and then if not working to start a new fresh OS install? This install came pre-loaded on a sd card. I just assumed it was ok.

bennuttall, 2021-07-19:
You could try what @bcroston suggested but I don't think that will help in this situation as you're using a stock Buster image.

Next, try installing a new image.

pigscoven, 2021-07-19:
I decided to do the complete fresh install. I placed this install on a different SD card because if it didn't work, I would still have the original install. I was still not able to use the above command with gpiozero, however I found someone else looking for help with their code on the pi forum and this code is working. I am not sure if the pin # is correct. It just keeps running that is detecting motion. I also have not added the wait for motion command. I have tried two different pin # in the code to just to see what it would look like if I was using the wrong pin #. It still worked so I am confused . Code I am working on now is 

import RPi.GPIO as GPIO
import time

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
pir = (22)
GPIO.setup(pir, GPIO.IN)

print("Starting")
time.sleep(1)
print("Ready")

while True:
    print("motion detected")
    time.sleep(1)

I really do appreciate your help!! I want to thank you! @bennuttall  and @bcroston

lurch, 2021-07-19:
`import RPi.GPIO as GPIO` is using the "base" [RPi.GPIO](https://sourceforge.net/projects/raspberry-gpio-python/) library, which doesn't have the extra complexity (like support for [PIR sensors](https://gpiozero.readthedocs.io/en/stable/api_input.html#gpiozero.MotionSensor) ) that [GpioZero](https://github.com/gpiozero/gpiozero) does. GpioZero "sits on top of" RPi.GPIO.

pigscoven, 2021-07-19:
In your opinion, I should try learning zero and get that right because it has more functions than GpioZero and there’s support and not for Rpi.GPIO? I do want to learn the newest of the two and it sounds like zero is the way to go. I appreciate your help. I’m going to spend some time reading the library for zero and see what I am doing wrong. 

Tina M Davis

lurch, 2021-07-19:
Yes, GpioZero was specifically designed to be easy to use for beginners. See https://gpiozero.readthedocs.io/en/stable/faq.html#why-is-it-called-gpio-zero-does-it-only-work-on-pi-zero

> I was still not able to use the above command with gpiozero

What do each of these commands say:
```
cat /proc/device-tree/system/linux,revision | hexdump -C

cat /proc/cpuinfo

apt policy python3-gpiozero python3-rpi.gpio
```
?

bennuttall, 2021-07-19:
You might want to read through some pages of the documentation here: https://gpiozero.readthedocs.io/en/stable/

Maybe look at a book like this too: https://magpi.raspberrypi.org/books/essentials-gpio-zero-v1

pigscoven, 2021-07-20:
@lurch cat is used to display a files content. I think of it as a catalog with extra information to read about a particular file. proc, I am not so sure but judging from the rest of the line, I am venturing to guess provide. device tree-tree is the, for lack of a word, its how one file starts at home and trickles down all the other files in till I get to the one I need. system-linux is the system I am using, revision means the revised version. hexdump is a display of the file but in hexadecimal, decimal, etc. -c, i have to think a minute here, if I am right, that command clears the contents. the first line cat /proc/device-tree/sytsem/linux, version | hexdump -c is the command for show me the cat providing information in the system tree on the linux version I have in computer code and then clear?
The second one is go to file and provide me the cpu information. The last one says to use the software to show me the policy of python3-rpi.gpio and to get there you start with python3-gpiozero. 
Correct me if needed. I do tend to get some mixed up. 
@bennuttall I have actually saved a lot of pages on my computer to read. I did not see the book so I will retrieve that. I was able to get the basic commands working, then I added some loops. I am not finished with my project, however I believe I will be god to go. I did install the information files that was on a webpage I am thinking you posted right after I installed a fresh copy. I used that last night to read and figured out some more commands. I really do appreciate all the help you  have been giving me. I have not used this information since la few years ago while in school. I have a couple of classes coming up so I decided to create a project for Halloween to freshen up my memory. Python is one I have not used. This will be a learning experience.

lurch, 2021-07-20:
@pigscoven I'm afraid you misunderstood me; I wasn't asking you to explain to me what those commands do, I was asking you to run them on your Raspberry Pi and tell me what they output, so that I could try and help you diagnose why you were getting the "Not running on a RPi" error :wink: 

But maybe that's not necessary now, if you were "able to get the basic commands working" ? I'm afraid I can't tell from your last reply if you're still having trouble getting GpioZero working or not?

pigscoven, 2021-07-20:
No. I am not having trouble now. I am sure re-installing a fresh copy of pi did the trick. Thank you. 

Tina M Davis

closed 2021-07-21 by bennuttall