gpiozero/gpiozero #983: Unncessary/unintended print() statement on LoadAverage.load_average ?

thiagosfsilva, 2021-07-22:
Hi there,

I noticed that on `internal_devices.py` the `load_average` property of the `LoadAverage` class seems to unnecessarily print the representation of the opened file as well as the stored value:

```python
Python 3.7.3 (default, Jan 22 2021, 20:04:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import CPUTemperature, DiskUsage, LoadAverage
>>> loadCPU = LoadAverage(min_load_average=0, max_load_average=1, minutes=1)    
>>> print(loadCPU.load_average)
<_io.TextIOWrapper name='/proc/loadavg' mode='r' encoding='UTF-8'>
0.76
>>> 
```

That doesn't seem to be the intended behavior, since similar parameters for other classes do not behave this way:
```python
>>> diskUsagePercent = DiskUsage()
>>> print(diskUsagePercent.usage)
14.494867817330109
>>> 
```

This is the current implementation of `LoadAverage.load_average`, starting on line 434 of `internal_devices.py`, with the referred print statement after opening the `load_average_file`:

```python

    @property
    def load_average(self):
        """
        Returns the current load average.
        """
        with io.open(self.load_average_file, 'r') as f:
            print(repr(f))
            file_columns = f.read().strip().split()
            return float(file_columns[self._load_average_file_column])
```

Thank you for looking into it.

lurch, 2021-07-22:
Oops, indeed that looks like left-over debugging code. Do you want to submit a PR?

thiagosfsilva, 2021-07-22:
I thought so but felt it was better to double-check first. PS submitted.

closed 2021-07-22 by waveform80

mahoefer, 2022-04-24:
this is still present in version (1.6.2-1) when installing on debian 11 with apt as python3-gpiozero

lurch, 2022-04-24:
> this is still present in version (1.6.2-1) when installing on debian 11 (python 3.9.2-3) with apt as python3-gpiozero

Yup, the [last published version](https://github.com/gpiozero/gpiozero/tags) of GPIO Zero was v1.6.2 in March 2021, and this particular bug was fixed in July 2021 (and so hasn't yet appeared in a published version).

mahoefer, 2022-04-25:
> Yup, the [last published version](https://github.com/gpiozero/gpiozero/tags) of GPIO Zero was v1.6.2 in March 2021, and this particular bug was fixed in July 2021 (and so hasn't yet appeared in a [...]

Thanks. I checked the releases log and didn't see anything. I didn't think to check the tags.... :/

commit 05e39eb3 (2021-07-22): Merge pull request #984 from thiagosfsilva/fix_load_average

closes #983 - removed print statement left from debugging
commit 61a3055a (2021-07-22): closes #983 - removed print statement left from debugging