gpiozero/gpiozero #985: Blink method has low accuracy

yagui, 2021-08-12:
Operating system: Raspbian Buster
Python version: 3.7
Pi model: e.g. Pi 4 Model B
GPIO Zero version: e.g. 1.6.1
Pin factory used: e.g. RPiGPIO

Hi,
I'm using the blink method from a DIgitalOutputDevice and I intend to do a 5 Hz square signal. Although I used on_time=.1 and off_time=.1, instead of getting a 200 ms pulse I'm getting something similar to 250 ms. including some jitter. I don't mind about the jitter but I was expecting to have something closer to 200 ms.

lurch, 2021-08-12:
If you have strict timing requirements, Python running under a multitasking OS is perhaps a poor choice.
Maybe https://www.raspberrypi.org/documentation/rp2040/getting-started/#board-specifications would be a better option? :shrug:

yagui, 2021-08-13:
I don't have strict timing requirements and I understand what you are referring to. I just though that because this square pulse is not that demanding, there could be a systematic delay in the library but I was wrong.
After doing more testing, I found out that gpiozero library works ok by it's own, with a jitter no bigger than 1 ms.
But because I am using pygame and pygame-menu, there are a lot of threads running and that is interfering with the blink thread.
Thanks

closed 2021-08-13 by yagui