gpiozero/gpiozero #991: Failed to add edge detection error

EDIflyer, 2021-09-05:
Operating system: Raspbian Buster
Python version: 3.7.3
Pi model: Pi Zero W
GPIO Zero version: 1.6.2
Pin factory used: RPiGPIO (`<gpiozero.pins.rpigpio.RPiGPIOFactory object at 0xb64324b0>`)

If I just press the touch pad occasionally it seems to be OK, however if I press for a longer time (such that a rapid number of detection events happen) then I get the following error.  If I add a time.sleep(0.25) this seems to help quite a bit (still testing to see if it resolves completely or not).  So I'm not sure if it's something to do with the more rapid updating of the GPIO pin causing the problem?

Error message:
> Traceback (most recent call last):
>   File "./touchtest.py", line 69, in <module>
>     touchtest()
>   File "./touchtest.py", line 63, in touchtest
>     touchalert = Button(4, pull_up=True)
>   File "/usr/local/lib/python3.7/dist-packages/gpiozero/devices.py", line 108, in __call__
>     self = super(GPIOMeta, cls).__call__(*args, **kwargs)
>   File "/usr/local/lib/python3.7/dist-packages/gpiozero/input_devices.py", line 416, in __init__
>     bounce_time=bounce_time, pin_factory=pin_factory)
>   File "/usr/local/lib/python3.7/dist-packages/gpiozero/mixins.py", line 433, in __init__
>     super(HoldMixin, self).__init__(*args, **kwargs)
>   File "/usr/local/lib/python3.7/dist-packages/gpiozero/input_devices.py", line 171, in __init__
>     self.pin.when_changed = self._pin_changed
>   File "/usr/local/lib/python3.7/dist-packages/gpiozero/pins/__init__.py", line 432, in <lambda>
>     lambda self, value: self._set_when_changed(value),
>   File "/usr/local/lib/python3.7/dist-packages/gpiozero/pins/pi.py", line 319, in _set_when_changed
>     self._enable_event_detect()
>   File "/usr/local/lib/python3.7/dist-packages/gpiozero/pins/rpigpio.py", line 229, in _enable_event_detect
>     bouncetime=self._bounce)
> RuntimeError: Failed to add edge detection

```python
def read_keystatus():
    key = i2cbus.read_byte_data(i2caddress, KEYSTATUS)
    print("Raw key: ", key)
    if key == 8: key_pressed = "Centre"
    if key == 4: key_pressed = "Left"
    if key == 2: key_pressed = "Right"
    if key == 0: key_pressed = "Nil"
    print("Key pressed: ", key_pressed)

def touchtest():
    t = time.localtime()
    current_time = time.strftime("%H:%M:%S", t)
    touchalert = Button(4, pull_up=True)
    touchalert.wait_for_press()
    print("Touch message waiting! ", current_time)
    read_keystatus()
```
SPI & I2C interfaces are enabled. Single-wire is not.
I wasn't sure if linked to #687 at all?  This was from years ago and seemed to be fixed though?

bennuttall, 2021-09-06:
You need to register the button once (outside of the function), like so:

```python
touch_alert = Button(4, pull_up=True)

def touch_test():
    t = time.localtime()
    current_time = time.strftime("%H:%M:%S", t)
    touchalert.wait_for_press()
    print("Touch message waiting! ", current_time)
    read_keystatus()
```

EDIflyer, 2021-09-06:
Doh! Well spotted @bennuttall - thanks 👍  That did indeed do the trick!
Next challenge is to try to sort [out an I2C issue I've having](https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=319084), but that's not a GPIOZero issue!

closed 2021-09-06 by EDIflyer

lurch, 2021-09-09:
Slightly OffTopic: Seems a bit odd to read the `current_time`, then `wait_for_press`, and then print `current_time` _after_ the button-press has (eventually) happened? :shrug:

EDIflyer, 2021-09-09:
Thanks @lurch - yep that's a fair point but TBH I wasn't actually bothered what the time itself was, I was just trying to quickly differentiate between the presses and test how quickly it could iterate so was really just for a bit of working things out. Cheers though :)