gpiozero/gpiozero #994: unable to locate Pi revision in /proc/cpuinfo

baterflyrity, 2021-09-24:
Raspberry Pi 4, Debian 11, gpiozero 1.4.1.

When i try to bind button to pin i receive this exception: "unable to locate Pi revision in /proc/cpuinfo". What should i do to resolve this issue? Thanks in advance.

`Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 71, in pin
    pin = self.pins[n]
KeyError: 2

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 95, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 302, in __init__
    super(Button, self).__init__(
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 338, in __init__
    super(HoldMixin, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 94, in __init__
    super(DigitalInputDevice, self).__init__(
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 164, in __init__
    super(EventsMixin, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 46, in __init__
    super(InputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 375, in __init__
    pin = self.pin_factory.pin(pin)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 73, in pin
    pin = self.pin_class(self, n)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 94, in __init__
    super(RPiGPIOPin, self).__init__(factory, number)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 241, in __init__
    factory.pi_info.physical_pin(repr(self))
  File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 130, in <lambda>
    lambda self: self._get_pi_info(),
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 90, in _get_pi_info
    self._info = pi_info(self._get_revision())
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 67, in _get_revision
    raise PinUnknownPi('unable to locate Pi revision in /proc/cpuinfo')
gpiozero.exc.PinUnknownPi: unable to locate Pi revision in /proc/cpuinfo
`

lurch, 2021-09-24:
gpiozero 1.4.1 is very old... https://github.com/gpiozero/gpiozero/releases
Try using a newer version? https://gpiozero.readthedocs.io/en/stable/installing.html

baterflyrity, 2021-09-24:
> 
> 
> gpiozero 1.4.1 is very old... https://github.com/gpiozero/gpiozero/releases
> Try using a newer version? https://gpiozero.readthedocs.io/en/stable/installing.html

Thanks a lot. Somehow my new and fresh debian 11 was shipped with old 1.4.1 version. But using pip instead of apt i solved this issue (`python3 -m pip install --upgrade gpiozero`).

closed 2021-09-24 by baterflyrity