gpiozero/gpiozero #995: Mmap of GPIO registers failed

baterflyrity, 2021-09-24:
Operating system: debian 11 x64
Python version: 3.9
Pi model: 3 B+
GPIO Zero version: 1.6.2
Pin factory used: RPiGPIO (0.7.1), pigpio (1.78), native.

When binding (running ptyhon as root) button **mmap error** occurs independently of pin factory.

```python
>>> from gpiozero import Button
>>> button = Button(2)
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/gpiozero/pins/pi.py", line 101, in pin
    pin = self.pins[n]
KeyError: 24

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/gpiozero/devices.py", line 108, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/gpiozero/input_devices.py", line 414, in __init__
    super(Button, self).__init__(
  File "/usr/local/lib/python3.9/dist-packages/gpiozero/mixins.py", line 433, in __init__
    super(HoldMixin, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/gpiozero/input_devices.py", line 165, in __init__
    super(DigitalInputDevice, self).__init__(
  File "/usr/local/lib/python3.9/dist-packages/gpiozero/mixins.py", line 218, in __init__
    super(EventsMixin, self).__init__(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/gpiozero/input_devices.py", line 83, in __init__
    super(InputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/usr/local/lib/python3.9/dist-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
  File "/usr/local/lib/python3.9/dist-packages/gpiozero/pins/pi.py", line 103, in pin
    pin = self.pin_class(self, n)
  File "/usr/local/lib/python3.9/dist-packages/gpiozero/pins/rpigpio.py", line 111, in __init__
    GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Mmap of GPIO registers failed
```

lurch, 2021-09-24:
> Operating system: debian 11 x64
> Pi model: 3 B+

Typo? The Pi isn't able to run x64 OSes :wink: 

> RuntimeError: Mmap of GPIO registers failed

Does (the permissions stuff in) #837 help?

baterflyrity, 2021-09-25:
> Typo? The Pi isn't able to run x64 OSes 😉
Sry, ofc its [x32](https://wiki.debian.org/ArmHardFloatPort).

baterflyrity, 2021-09-25:
> 
> 
> > Typo? The Pi isn't able to run x64 OSes 😉
> 
> Sry, ofc its [x32](https://wiki.debian.org/ArmHardFloatPort).

But i get:
```bash
>>>uname -a
Linux operator 5.10.0-8-arm64 #1 SMP Debian 5.10.46-4 (2021-08-03) aarch64 GNU/Linux
```

lurch, 2021-09-25:
Going a bit off-topic, but the 'x' in x32 and x64 generally stands for x86 (i.e. the architecture used by Intel / AMD CPUs). The Pi uses ARM CPUs, which are generally referred to as armhf for 32-bit OSes, or aarch64 for 64-bit OSes. :computer: :slightly_smiling_face: (sorry for being pedantic)

baterflyrity, 2021-09-25:
@lurch thanks for link but now the problem seems to be in no gpiomem device:

```bash
root@debian:~# ls /dev/ | grep gpio
gpiochip0
gpiochip1
```

How can i add gpiomem device?

bennuttall, 2021-09-25:
You'd be better off using either Raspberry Pi OS or Ubuntu, as these both provide working GPIO interfaces. See https://waldorf.waveform.org.uk/2021/the-pins-they-are-a-changin.html for more info on how the different pin interfaces work.

closed 2021-09-25 by bennuttall