gpiozero/gpiozero #999: Add requirements(dependencies) to docs

baterflyrity, 2021-09-28:
According to this [thread](https://github.com/gpiozero/gpiozero/discussions/996#discussioncomment-1384366_) gpiozero requires preinstalled pin library but its not stated in docs.

Suggest to include backends requirements and maybe short instructions to their installation in documentation.

lurch, 2021-09-28:
There's a bit of documentation at https://gpiozero.readthedocs.io/en/stable/api_pins.html and https://gpiozero.readthedocs.io/en/stable/faq.html#why-do-i-get-pinfactoryfallback-warnings-when-i-import-gpiozero but you're right in that it wouldn't hurt to add a bit of extra info to https://gpiozero.readthedocs.io/en/stable/installing.html

baterflyrity, 2021-09-28:
@lurch Exactly, coz pip often installs/builds required dependencies hence there is a small confusion around gpiozero package.

p.s.
Also i consider it worse mentioning that aptitude can install old bugged versions of gpiozero.