gpiozero/gpiozero #1000: How to use gpiozero on RPi4 with Ubuntu 21.04, errored out NotImplemented

franva, 2021-10-02:
Operating system:  Ubuntu 21.04 Hirsute
Python version: e.g. 3.8.8
Pi model: e.g. Pi 4 Model B 4GB RAM
GPIO Zero version: e.g. 1.6.1
Pin factory used:  RPiGPIO

```python
# import os
# os.environ['GPIOZERO_PIN_FACTORY'] = os.environ.get('GPIOZERO_PIN_FACTORY', 'pigpio')

import time
# from gpiozero.pins.pigpio import PiFactory
from gpiozero import Motor, Device
#Device.pin_factory = PiFactory()
# from gpiozero.pins.native import  LocalPiFactory
# from gpiozero.pins.rpigpio import LocalPiFactory
# from gpiozero.pins.rpio import LocalPiFactory
# from gpiozero.pins.lgpio import LocalPiFactory

# ###
# drive motor: 0.5 - 1, turn motor : 0.2 -1
# ###
class Car:
    FORWARD = 'FORWARD'
    BACK = 'BACK'
    LEFT = 'LEFT'
    RIGHT = 'RIGHT'
    CURVE_LEFT = 'CURVE_LEFT'
    CURVE_RIGHT = 'CURVE_RIGHT'

    def __init__(self) -> None:
        self.motor_drive = Motor(6,13)
        self.motor_turn = Motor(12,16)
        self.turn_low_power = 0.2
        self.turn_high_power = 0.8
        self.step = (self.turn_high_power - self.turn_low_power) / 100.0
```
I'm trying to drive the motors from my RPi4. On the RPi4 4GB RAM, I installed Ubuntu 21.04.
The code, it used to be working on RPi Buster OS, but once I changed the OS to Ubuntu 21.04, it stopped working.

Searched and found that I need to set up environment variables e.g. GPIOZERO_PIN_FACTORY and also need to install gpiozero and pigpio and I pip installed them and tried to use code to set the pin factory as you could see in the commented code etc.

But none of them works. An error shows: NotImplemented on this line: self.motor_drive = Motor(6,13)

[image]

Could someone show me how to make it work on Ubuntu 21.04?

franva, 2021-10-02:
Update 1:

After upgrading my Ubuntu to 21.10, it still does not work.

More details here:

https://raspberrypi.stackexchange.com/questions/132447/how-to-use-gpiozero-on-rpi4-with-ubuntu-21-04-errored-out-notimplemented

lurch, 2021-10-02:
See more info in https://github.com/gpiozero/gpiozero/search?q=ubuntu&type=issues and https://github.com/gpiozero/gpiozero/discussions/996

franva, 2021-10-04:
yep, I found there is no solution to use gpio on Ubuntu so far. So will switch back to the 32 bit RPi OS.....

It would be lovely to see that the gpiozero to be supported on Ubuntu's 64bit OS to utilize more system resources