gpiozero/gpiozero #1005: Rotary Encoder no output or rotation detected

Shinora, 2021-11-16:
Operating system:  Raspbian OS lite
Python version:  3.9
Pi model: Pi Zero W
Pin factory used:  RPiGPIO

I am having no output with my code, nothing from the rotary encoder, the fonction wait_for_rotate() just waits for an infinite amount of time and rotor.value() keeps sending 0

```python
from gpiozero import RotaryEncoder
import time

rotor = RotaryEncoder(18, 17)
while True:
    print(rotor.value())
    time.sleep(1)

```
If i may add that this piece of code works fine for me : 

```python
from RPi import GPIO
from time import sleep

def button_callback(channel):
    print("Button pressed")

clk = 17
dt = 18
btn = 15

GPIO.setmode(GPIO.BCM)
GPIO.setup(clk, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.setup(dt, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.setup(btn, GPIO.IN, pull_up_down=GPIO.PUD_DOWN)
GPIO.add_event_detect(15,GPIO.RISING,callback=button_callback)
counter = 0
clkLastState = GPIO.input(clk)

try:

    while True:
        clkState = GPIO.input(clk)
        dtState = GPIO.input(dt)
        if clkState != clkLastState:
            if dtState != clkState:
                counter += 1
            else:
                counter -= 1
            print(counter)
            clkLastState = clkState
            sleep(0.01)
finally:
        GPIO.cleanup()```

Shinora, 2021-12-23:
I have been searching a lot about this without any anwser. After hours i just tried to connect the rotary encoder to the same pins that are mentionned in the documentation, and it worked ! I don't get what was wrong with my pins, maybe the issue was from the rpi but it works with pin A and B set to 21 and 20 !

closed 2021-12-23 by Shinora

lurch, 2021-12-24:
Weird - perhaps you've accidentally damaged those pins on your Pi? (which might happen if you accidentally connect them to 5V)

Shinora, 2021-12-24:
> Weird - perhaps you've accidentally damaged those pins on your Pi? (which might happen if you accidentally connect them to 5V)

Honestly i have no idea, i tried a few different pins and none worked with gpiozero but it worked if i coded it from scratch with GPIO from RPi, like you can see above. 
I am happy that it works now but i don't understand what happened, like not at all.