gpiozero/gpiozero #1007: Button causes high cpu usage

himijendrix24, 2021-12-07:
Operating system: e.g. Raspbian Bullseye
Python version: e.g. 3.9.2
Pi model: Pi 3 Model B+
GPIO Zero version: e.g. 1.6.2
Pin factory used: e.g. RPiGPIO

Used Shields: 
Joy-IT StromPiV3 (using GPIO 14, 15)
Hifiberry DAC+ ADC (using GPIO 0,1,2,3,18,19,20,21)

I use two buttons on GPIO 22 and GPIO23 and noted some really strange behaviour during programming.
When polling the pins with the following sample script I get very unstable results (flickering between 0 and 1).
At first I thought it was caused by the high internal pull-up resistance and added some 4.7k pull ups.
Then I noticed that as soon as I define the gpiozero pins python3's cpu usage goes to ~ 100% and stays there.
The process 'kworker/0:0+events' also goes to 100%.

Running as sudo does not change anything.

Adding a debounce time does not change anything.

Both shields (hifiberry and strompi) are configured using their manuals and work as expected.
GND is properly connected, even through the stack of shields.

```python
from gpiozero import Button
import time 

btn = Button(23, pull_up=True)

while True:
    print(btn.value)
    time.sleep(0.001)
```

When calling `btn.close()` I randomly get a `Segmentation fault` python crash.

Any idea what could be the reason?

lurch, 2021-12-07:
Sounds suspiciously similar to #843 ?

Also, do things work better if you increase the `sleep()` time to something more than 1 millisecond?

bennuttall, 2021-12-07:
Does the following work for you?

```python
from gpiozero import Button
import time 

btn = Button(23, pull_up=True)

while True:
    btn.wait_for_press()
    print("Pressed")
    btn.wait_for_release()
    print("Released)
```

himijendrix24, 2021-12-07:
As soon as `btn = Button(23, pull_up=True)` is called the CPU usage gets stuck. 
Even after quitting _kworker/0:0+events_ hangs at 100% for a longer time.

Tomorrow I am going to recheck everything first without the shields and then on a new raspbian image.
It seems like a classical hardware problem with a noisy input pin, but on the other hand it is a more a software problem.

I also tried different pin factories which did not change anything.

bennuttall, 2021-12-07:
What about just using RPi.GPIO without gpiozero? Do you get the same or different behaviour?

himijendrix24, 2021-12-08:
After removing the Strompi and Hifiberry shields it all worked as expected and also gpiozero works well.

With the shields only RPi.GPIO directly it works without clogging the cpu.

However the values still flickered as soon as a wire (~ 1.5m, 0.5mm² to the switch) was connected.
This is very unusual with a 1.2k pull-up resistor. 
I changed wires, soldered directly to the raspberry pins and re-checked everything with the mulitmeter.
No real improvements.

I had to add a 4.7uF capacitors accross the switch and directly to the raspberry pin, which now produces stable values.

gpiozero still causes high cpu usage when closing the switch, but NOT when shorting the pin directly to GND with a short wire. 

So its definitely some really weird hardware problem caused by the shields and the long wire (... although 1.5m are not that long) inducing noise to the pin and causing to many state changes.

lurch, 2021-12-08:
Are you in an "electrically noisy" environment? (motors, relays switching on and off, etc.) A 1.5m wire can act as an "antenna" and pick up any stray [EMI](https://en.wikipedia.org/wiki/Electromagnetic_interference), leading to the noisy values you're seeing.
I guess you could either try using a shorter wire and moving the Pi nearer to the switch, or maybe you could "shield" the wire by [wrapping it in tinfoil](https://www.stuffaboutcode.com/2014/01/raspberry-pi-gps-helmet-cam.html) or something? :man_shrugging:

craigerl, 2023-11-28:
```
from gpiozero import Button
button24 = Button(24)
button23 = Button(23)
button23.when_pressed = button_callback_23
```

Just instantiating the button objects drives the cpu up (2% on pi5,  about 7% on a pi4).

Using RPI.gpio works as expected with no CPU load.

Is there a way we can poll the buttons directly with gpiozero, putting in a reasonable sleep?

Raspberry Pi5, watching buttons is the top cpu use,
```
top - 17:16:48 up 2 min,  1 user,  load average: 0.19, 0.29, 0.13
Tasks: 169 total,   1 running, 168 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0.2 us,  0.4 sy,  0.0 ni, 99.4 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st 
MiB Mem :   3993.2 total,   3633.7 free,    227.9 used,    216.8 buff/cache     
MiB Swap:    100.0 total,    100.0 free,      0.0 used.   3765.3 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                                        
 1303 pi        20   0   67100  14680   6044 S   2.0   0.4   0:00.39 digibuttons.gpi
```

gpiozero threads,
```
$ ps -T
  PID  SPID TTY          TIME CMD
 1201  1201 pts/1    00:00:00 bash
 1303  1303 pts/1    00:00:01 digibuttons.gpi
 1303  1304 pts/1    00:00:00 digibuttons.gpi
 1303  1305 pts/1    00:00:00 digibuttons.gpi
 1303  1306 pts/1    00:00:00 digibuttons.gpi
 1303  1308 pts/1    00:00:00 digibuttons.gpi
 1303  1310 pts/1    00:00:00 digibuttons.gpi
```

bennuttall, 2023-11-28:
Use one of the alternative approaches: https://gpiozero.readthedocs.io/en/latest/faq.html#what-s-the-difference-between-when-pressed-is-pressed-and-wait-for-press

craigerl, 2023-12-01:
Can you provide a specific example?  using is_pressed, when_pressed, etc, makes no difference.  The CPU load shoots up as soon as you define the button.  

I'm talking about a Pi Zero here, every cycle counts.  With gpiozero, monitoring a pin takes as much CPU as an audio encoder.  That's my entire budget.   ... otherwise, RPI.gpio wins every time here.

craigerl, 2023-12-01:
I didn't realize this thread was several years old, so I'm going with nobody cares.  For those who cant afford to burn half a cpu core on a PiZero with gpiozero, just to monitor a button this is a gpiod  implementation that takes virtually no cpu usage... 

```
#!/usr/bin/python3
import select
import gpiod
import time
import threading
from signal import pause

def button_callback_23(number):
    print("Button ",   number,  "  pressed.")
    return(0)

def thread23():
   lastcalltime = 0
   chip = gpiod.Chip('/dev/gpiochip4')
   line23 = chip.get_line(23)
   line23.request(consumer="GPIN", type=gpiod.LINE_REQ_EV_FALLING_EDGE, flags=gpiod.LINE_REQ_FLAG_BIAS_PULL_UP)
   fd23 = line23.event_get_fd()
   poll = select.poll()
   poll.register(fd23)
   while True:
      event = line23.event_read()
      if ((time.time() - lastcalltime) >= 1):  # one second debounce
         lastcalltime = time.time()
         button_callback_23(23)

t23 = threading.Thread(target=thread23)
t23.start()

pause()

```

lurch, 2023-12-02:
Note that `Button` inherits from a bunch of other classes https://gpiozero.readthedocs.io/en/latest/api_input.html#base-classes each of which adds more sophistication on top of earlier layers. So if `Button` is too slow for you, you might want to try `DigitalInputDevice` or `InputDevice` to see if they give you better performance?
I also see that in your gpiod-based code you're using a one-second debounce, but in your earlier gpiozero-based code you didn't provide a `bounce_time` value to the Buttons.

rasgeek, 2024-03-01:
Hi @lurch thanks for this clarification I used DigitalInputDevice in the below code and it improved my cpu usage on a raspberry B+, I am trying to log/count  pulses / button presses to database. 

from gpiozero import DigitalInputDevice
from datetime import datetime
import sqlite3

# Define GPIO pins for Meter In and Meter Out
meter_in_PIN = 2
meter_out_PIN = 3

# Get the MAC address
mac_address = 9336671
status = 1
current_time = datetime.now().strftime('%Y-%m-%d %H:%M:%S')

# Connect to the SQLite database
conn = sqlite3.connect('meter_log.db', check_same_thread=False)
c = conn.cursor()

# Create a table to store button presses for Meter In if it doesn't exist
c.execute('''CREATE TABLE IF NOT EXISTS meter_log
             (id INTEGER PRIMARY KEY AUTOINCREMENT,
             transact TEXT,
             date_time TEXT)''')

# Create a table to store device information if it doesn't exist
c.execute('''CREATE TABLE IF NOT EXISTS meter
             (id INTEGER PRIMARY KEY AUTOINCREMENT,
             meter_id TEXT,
             status INTEGER,
             sync_time TEXT)''')

# Insert device information into meter table if it doesn't exist
c.execute("INSERT OR IGNORE INTO meter (meter_id,status,sync_time) VALUES (?, ?, ?)", (mac_address, status, current_time))
conn.commit()

# Function to log button press
def log_button_press(button_name):
    # Get current date and time
    current_time = datetime.now().strftime('%Y-%m-%d %H:%M:%S')

    # # Log button press to debug table
    # c.execute("INSERT INTO meter_log (transact, date_time) VALUES (?, ?)", (current_time, button_name))
    # conn.commit()

# Initialize DigitalInputDevice objects for Meter In and Meter Out
meter_in = DigitalInputDevice(meter_in_PIN, pull_up=True)
meter_out = DigitalInputDevice(meter_out_PIN, pull_up=True)

# Function to handle button press events
def button_pressed(button):
    if button == meter_in:
        button_name = "Meter In"
    elif button == meter_out:
        button_name = "Meter Out"
    else:
        return

    # Log button press
    log_button_press(button_name)
    print(f"{button_name} pressed")

# Attach event handlers for button press events
meter_in.when_activated = lambda: button_pressed(meter_in)
meter_out.when_activated = lambda: button_pressed(meter_out)

# CPU reduction test
# Keep the script running
try:
    while True:
        pass
except KeyboardInterrupt:
    conn.close()

lurch, 2024-03-01:
Can you put your code inside triple-backticks please, to stop GitHub messing up the formatting? And if you're doing a database-write on every button-press, then it's no surprise that it's slow! :laughing:

omya3qno, 2025-01-22:
Problem confirmed with a Raspberry Pi Zero W Rev 1.1, python 3.11.2, python3-gpiozero 2.0.1-0+rpt1. The python process running the code below uses about 12% CPU, even when no triggers arrive.

```
from gpiozero import DigitalInputDevice
import signal
def yo(): print('yo')
b = DigitalInputDevice(pin=21)
b.when_activated = yo
signal.pause()
```

Using ```strace -ff``` on the python process shows plenty activity, so something is nearly spin-waiting

```
[pid  6862] ppoll([{fd=6, events=POLLIN|POLLPRI}], 1, {tv_sec=0, tv_nsec=500000}, NULL, 8) = 0 (Timeout)
[pid  6862] clock_gettime64(CLOCK_MONOTONIC, {tv_sec=189923, tv_nsec=950141717}) = 0
[pid  6862] ppoll([{fd=6, events=POLLIN|POLLPRI}], 1, {tv_sec=0, tv_nsec=500000}, NULL, 8 <unfinished ...>
[pid  6861] <... clock_nanosleep resumed>0xb57fed2c) = 0
[pid  6862] <... ppoll resumed>)        = 0 (Timeout)
[pid  6861] clock_nanosleep(CLOCK_MONOTONIC, TIMER_ABSTIME, {tv_sec=189923, tv_nsec=970158227},  <unfinished ...>
[pid  6862] clock_gettime64(CLOCK_MONOTONIC, {tv_sec=189923, tv_nsec=953279713}) = 0
[pid  6862] ppoll([{fd=6, events=POLLIN|POLLPRI}], 1, {tv_sec=0, tv_nsec=500000}, NULL, 8) = 0 (Timeout)
[pid  6862] clock_gettime64(CLOCK_MONOTONIC, {tv_sec=189923, tv_nsec=956074710}) = 0
```

lurch, 2025-01-23:
The comments in #1165 suggest that the `gpiod` pin library in #1185 might be more efficient. Would you like to give it a test on your Raspberry Pi Zero W?