gpiozero/gpiozero #1008: DigitalInputPin does not detect signals from hall sensors

thehamzan6, 2021-12-25:
Hi everyone,

for some reason DigitalInputPin does not detect any signal from my hall sensor attached to a motor. Reading the input using GPIO, raspberry pi does indeed detect an input.
`
from gpiozero import *
from time import sleep
class QuadratureEncoder(object):
    """
    A simple quadrature encoder class
    Note - this class does not determine direction
    """
    def __init__(self, pin_a, pin_b):
        self._value = 0

        encoder_a = DigitalInputDevice(pin_a)
        encoder_a.when_activated = self._increment
        encoder_a.when_deactivated = self._increment

        encoder_b = DigitalInputDevice(pin_b)
        encoder_b.when_activated = self._increment
        encoder_b.when_deactivated = self._increment
        
    def reset(self):
        self._value = 0

    def _increment(self):
        self._value += 1

    @property
    def value(self):
        return self._value

r = Robot(left=(6,5,12), right=(26,19,13)) 
e1 = QuadratureEncoder(4, 17)
e2 = QuadratureEncoder(27, 22)

r.value = (1,1)
#r.value = (0,0)
while True:
    print("e1 {} e2 {}".format(e1.value, e2.value))
    sleep(0.1)
`

I am using this code, and I have 0 and 0 as the output. Anyone have any ideas?

lurch, 2021-12-26:
What pin-~~library~~factory are you using? https://gpiozero.readthedocs.io/en/stable/api_pins.html

> Reading the input using GPIO, raspberry pi does indeed detect an input.

Can you show the equivalent code that _does_ work with native RPi.GPIO?

Can you try rotating the motors by hand, to see if the "edges" are too fast for GpioZero to detect?

thehamzan6, 2021-12-26:
Thank you for your response, 

```
import RPi.GPIO as GPIO
from time import sleep

#definition of GPIO pins corresponding to motor encoder
HallS1 = 22#4
HallS2 = 27#17
# counter to count ticks or rotations
counter = 0
# The state which is 1 or 0
aState = 0
aLastState = 0

#setup
GPIO.setmode(GPIO.BCM)
GPIO.setup(HallS1, GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.setup(HallS2, GPIO.IN)
#GPIO.setup(motorBHall1, GPIO.IN)
#GPIO.setup(motorBHall2, GPIO.IN)

#Read the initial state of Sensor 1
aLastState = GPIO.input(HallS1)

while True:
    aState = GPIO.input(HallS1)
    # If previous and current sate != each other, a pulse occured 
    if aState != aLastState:
        # If Hall2 != Hall1, it is encoder is rotating clockwise
        if GPIO.input(HallS2) != aState:
            counter += 1
        else:
            counter -= 1
    print("Position: " + str(counter))
    sleep(0.01)
    
aLastState = aState
GPIO.cleanup()
```
This is the code I have used to monitor the pulses that works.

However even by rotating the motors manually, I still receive no increments when using the gpiozero library.

lurch, 2021-12-26:
I dunno if it helps, but I see that you have `GPIO.setup(HallS1, GPIO.IN, pull_up_down=GPIO.PUD_UP)` whereas [DigitalInputDevice](https://gpiozero.readthedocs.io/en/stable/api_input.html#digitalinputdevice) defaults to `pull_up=False` ?

thehamzan6, 2021-12-26:
Unfortunately it doesn't make a difference whether the pull up is true or false. I don't know what to do. Is there anyway to debug the digitalinputdevice?

lurch, 2021-12-26:
I guess you could just try wiring it (the DigitalInputDevice) up to a push-button, to try and debug why it's working in RPi.GPIO but not GpioZero? (i.e. get it working at a "simpler" level, before moving back to your hall sensors?)

thehamzan6, 2021-12-27:
Weird. I did a button test, DigitalInputDevice also cannot detect it.

```
from gpiozero import *
from time import sleep

counter = 0

def increment():
    global counter
    counter +=1

test = DigitalInputDevice(21)
test.when_activated = increment()
test.when_deactivated = increment()

while True:
    print("Position:" + str(counter))
    sleep(0.00001)
```

Outputs: Position: 2

And the output does not change whether I press the button or not.

However, with RPi.GPIO,
```
import RPi.GPIO as GPIO
from time import sleep

GPIO.setmode(GPIO.BCM)
GPIO.setup(21, GPIO.IN)

while True:
    if (GPIO.input(21) == 1):
        print("Btn pressed")
```

Outputs: Btn pressed

Every time I press the button it outputs.

lurch, 2021-12-27:
```
test.when_activated = increment()
```
That assigns the **return** value of the `increment()` function to `.when_activated`. You instead want to assign a function-reference, like:
```
test.when_activated = increment
```

thehamzan6, 2021-12-28:
That fixed it. Thanks.

closed 2021-12-28 by thehamzan6