gpiozero/gpiozero #1009: Issue with using DigitalOutputDevice and A4988Nema from RpiMotorLib in the same time

vovagorodok, 2022-01-03:
Operating system: Raspberry Pi OS Bullseye
Python version: 3.9.2
Pi model: Pi Zero
GPIO Zero version: e.g. 1.6.2
Pin factory used: RPiGPIO

RpiMotorLib uses RPiGPIO as backend. Pins 20 and 21 are used for motor control
1. One additional output pin used RPi.GPIO like
```python
from RPi import GPIO
class Magnet:
    DEFAULT_PIN = 16
    def __init__(self, pin = DEFAULT_PIN):
        self.__pin = pin
        GPIO.setmode(GPIO.BCM)
        GPIO.setup(self.__pin, GPIO.OUT)
        self.off()
    def on(self):
        GPIO.output(self.__pin, True)
    def off(self):
        GPIO.output(self.__pin, False)
```
2. All works fine and after end of application motor pins 20 and 21 has logic 0 as outputs
3. Change RPi.GPIO to DigitalOutputDevice:
```python
from gpiozero import DigitalOutputDevice
class Magnet:
    DEFAULT_PIN = 16
    def __init__(self, pin = DEFAULT_PIN):
        self.__pin = DigitalOutputDevice(pin, initial_value=False)
    def on(self):
        self.__pin.on()
    def off(self):
        self.__pin.off()
```
4. After end of application motor pins 20 and 21 has some noise instead logic state. And motor make danger random moves

DigitalOutputDevice pin 16 has any influence to pins 20,21? Or method like GPIO.cleanup() is called in background?

lurch, 2022-01-03:
I dunno if this might have anything to do with e.g. #707 ?

Also, can you please show the code where GPIO pins 20 and 21 are used? And you can use the [raspi-gpio command](https://github.com/RPi-Distro/raspi-gpio) in Raspberry Pi OS to see what "mode" the GPIO pins are in, which might be useful for debugging this.

vovagorodok, 2022-01-03:
Just simple:
```python
from RpiMotorLib import RpiMotorLib
from magnet import Magnet

m = Magnet()
motor = RpiMotorLib.A4988Nema(20, 21, (-1, -1, -1))
motor.motor_go(clockwise = True, steptype = "Full", steps = 100, stepdelay = .001, verbose = False, initdelay = 0)
m.on()
motor.motor_go(clockwise = True, steptype = "Full", steps = 100, stepdelay = .001, verbose = False, initdelay = 0)
m.off()
```

vovagorodok, 2022-01-03:
After running with RPi.GPIO:
```bash
pi@prototype:~ $ raspi-gpio get 16,20,21
GPIO 16: level=0 fsel=1 func=OUTPUT
GPIO 20: level=0 fsel=1 func=OUTPUT
GPIO 21: level=0 fsel=1 func=OUTPUT
```

After running with DigitalOutputDevice
```bash
pi@prototype:~ $ raspi-gpio get 16,20,21
GPIO 16: level=0 fsel=0 func=INPUT
GPIO 20: level=1 fsel=0 func=INPUT
GPIO 21: level=1 fsel=0 func=INPUT
```

lurch, 2022-01-04:
Okay, so the difference is that RPi.GPIO leaves the pins as OUTPUT (i.e. actively being driven low), whereas GpioZero resets the pins to INPUT when it closes them, and they're obviously floating which is why they're picking up random values. Could you add some pull-down resistors to the pins, so that when they're inputs they have a "known value" but when they're outputs the Pi is able to "override" the value of the pull-down resistors?

vovagorodok, 2022-01-04:
Is used simple board for that:
https://allegro.pl/oferta/zestaw-cnc-grawerka-v4-shield-nano-3xstepstick-8781997007
Producer doesn't consider this case. And there no way to pull-down resistor without hacks and soldering
There is software solution to disable this deinitialization ans leave state of pins?
And it's ok that library change pins to output at the end, when at the beginning library get pins as input?

lurch, 2022-01-04:
Hmmm, I guess it's unfortunate that the board wasn't designed to cope with "floating inputs" :confused:  Although it appears to be designed for the Arduino rather than the Raspberry Pi, so I guess you must be using some kind of "adaptor cable" to connect it to your Pi? Perhaps you could add pull-down resistors at the Pi-end of the cable?

> There is software solution to disable this deinitialization and leave state of pins?

That's exactly what #707 is about.

> And it's ok that library change pins to output at the end, when at the beginning library get pins as input?

That's exactly what #803 is about.

It sounds like in the meantime your only real option is to just stick to RPi.GPIO rather than using GpioZero.

P.S. What does `raspi-gpio get 16,20,21` say when you first boot up the Pi, i.e. before running **any** GPIO scripts?

vovagorodok, 2022-01-04:
```bash
pi@prototype:~ $ raspi-gpio get 16,20,21
GPIO 16: level=0 fsel=0 func=INPUT
GPIO 20: level=0 fsel=0 func=INPUT
GPIO 21: level=0 fsel=0 func=INPUT
pi@prototype:~ $ raspi-gpio get 16,20,21
GPIO 16: level=0 fsel=0 func=INPUT
GPIO 20: level=0 fsel=0 func=INPUT
GPIO 21: level=0 fsel=0 func=INPUT
pi@prototype:~ $ raspi-gpio get 16,20,21
GPIO 16: level=0 fsel=0 func=INPUT
GPIO 20: level=0 fsel=0 func=INPUT
GPIO 21: level=0 fsel=0 func=INPUT
```

lurch, 2022-01-04:
Yup, RPi has both internal pull-up and pull-down resistors. On the RPi Zero you can **set** the pulls, but you can't **read** what the pulls have been set to. Only the RPi 4 and RPi 400 are able to both set and read the pulls.

So I guess one "workaround" might be to use the `raspi-gpio` command to explicitly set the internal pulls, after your GpioZero script has finished running?

vovagorodok, 2022-01-04:
Or if DigitalOutputDevice set pins to some values at the end, will be nice to set pins to default pin states? But what if at RPi are running two scripts with DigitalOutputDevice and first ends and set values to default? Than second script will have undefined behavior?
From my PoV will be ok if library set to default (or other defined value) only pins that was used by them and not touch other. If its possible :)

lurch, 2022-01-04:
This is all buried so deeply inside all the multiple layers of abstraction that @waveform80 has added (see https://gpiozero.readthedocs.io/en/stable/api_pins.html and https://gpiozero.readthedocs.io/en/stable/api_info.html and https://gpiozero.readthedocs.io/en/stable/api_output.html#base-classes ), that he's probably the only person that can offer any additional advice here.

vovagorodok, 2022-01-11:
Hi @waveform80 ,
can You comment?