gpiozero/gpiozero #1011: Virtual environment causes unusual PWM error with robot class

thehamzan6, 2022-01-07:
Operating system: Raspbian Buster
Python version: 3.7
Pi model: Pi 4 Model B
GPIO Zero version: e.g. 1.6.3

```python
import gpiozero
robot(LEFT=(6, 7, 12), RIGHT=(8,9,13))

```
https://stackoverflow.com/questions/66768739/gpiozero-exc-pinpwmunsupported-pwm-is-not-supported-on-pin-gpio7-raspberry-pi

I have gone through the same problem as the above article. Running the code outside a python virtual environment works, but not inside of it.

gpiozero.exc.PinPWMUnsupported: PWM is not supported on pin GPIO6

thehamzan6, 2022-01-07:
`pip install rpi.gpio`
inside venv.

closed 2022-01-07 by thehamzan6

lurch, 2022-01-07:
Yup, as documented at https://gpiozero.readthedocs.io/en/stable/development.html
And https://gpiozero.readthedocs.io/en/stable/installing.html says "To install GPIO Zero in a virtual environment, see the Development page.".