gpiozero/gpiozero #1014: MCP3201 Erros

dealerwrx, 2022-01-16:
Trying to use the MCP code for a MCP3201 chip to detect the movement of a Linear Actuator. The output from the gpiozero code never seems to change. the value is in decimal, but always the same [code below].
`
from gpiozero import Motor,MCP3201
motor = Motor(12,18)
pot = MCP3201(1)

print(pot.value)
motor.forward()
sleep(1)
motor.stop(1)
print(post.value)
`
When using a putty session with the RPI code [see below] (and another session to run the motor from the code below) the reading very from the 3.3 volts down to 0 like expected. 
`
import RPi.GPIO as GPIO
from time import sleep

GPIO.setmode(GPIO.BCM)
GPIO.setup(11, GPIO.OUT)
GPIO.setup(9,GPIO.IN)
GPIO.setup(8,GPIO.OUT)
clock = 11
miso = 9
cs = 8
GPIO.output(clock,True)
GPIO.output(cs,True)

while True:
    GPIO.output(cs,False)
    data = str()
    for i in range(16):
        GPIO.output(clock, False)
        if GPIO.input(miso):
            data += "1"
        else:
            data += "0"
        GPIO.output(clock, True)
    data = data.replace(' ','')[4:15]
    data = int(data,2)*(3.3/2045)
    data = str(data)
    print(data)
    GPIO.output(cs, True)
    sleep(.2)
`

Am I missing something in the gpiozero config?? Happy the share any other code that anyone needs.

lurch, 2022-01-17:
According to https://gpiozero.readthedocs.io/en/stable/api_spi.html#mcp3201, the arguments for the `MCP3201` class are `gpiozero.MCP3201(max_voltage=3.3, **spi_args)` - did you really mean to pass in `1` for the `max_voltage`?

dealerwrx, 2022-01-17:
even if I change that to 3.3 (on the 3.3 pin) or 5v, I only get 0.0001220852154804053 back from the pot.value