gpiozero/gpiozero #1015: Can't use two spi devices with hardware spi

slghb, 2022-01-19:
Trying to use two hardware SPI interfaces with different select_pins as shown below gives an `gpiozero.exc.GPIOPinInUse: pin 11 is already in use by SPI(port=0, device=0)` error.

```
from gpiozero import MCP3008

adc1 = MCP3008(device=0)
adc2 = MCP3008(device=1)
```

Here is the full error message:
```
/usr/lib/python3/dist-packages/gpiozero/pins/pi.py:156: SPISoftwareFallback: failed to initialize hardware SPI, falling back to software (error was: pin 9 is already in use by SPI(port=0, device=0))
  'software (error was: %s)' % str(e)))
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 82, in __call__
    self = cls._instances[key]()
KeyError: (11, 7)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 82, in __call__
    self = cls._instances[key]()
KeyError: (11, 10, 9)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "bug.py", line 5, in <module>
    adc2 = MCP3008(device=1)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 108, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 441, in __init__
    super(MCP3008, self).__init__(channel, differential, max_voltage, **spi_args)
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 297, in __init__
    **spi_args)
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 192, in __init__
    super(MCP3xxx, self).__init__(bits, max_voltage, **spi_args)
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 139, in __init__
    super(AnalogInputDevice, self).__init__(shared=True, **spi_args)
  File "/usr/lib/python3/dist-packages/gpiozero/spi_devices.py", line 45, in __init__
    self._spi = self.pin_factory.spi(**spi_args)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 158, in spi
    pin_factory=self, **spi_args)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 85, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/local.py", line 217, in __init__
    self._bus = SPISoftwareBus(clock_pin, mosi_pin, miso_pin)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 85, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/spi.py", line 35, in __init__
    self.clock = OutputDevice(clock_pin, active_high=True)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 108, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 83, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 85, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 548, in __init__
    self.pin_factory.reserve_pins(self, pin)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 92, in reserve_pins
    requester, *(self.pi_info.to_gpio(pin) for pin in pins))
  File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 78, in reserve_pins
    (pin, reserver))
gpiozero.exc.GPIOPinInUse: pin 11 is already in use by SPI(port=0, device=0)
```

I think the problem is the [LocalPiHardwareSPI class](https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/local.py#L116) is missing a `_conflicts_with` function, similar to the one in the [SPISoftware class](https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/spi.py#L53) or the [PiGPIOHardwareSPI class](https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/pigpio.py#L357).

Operating system: Raspbian Buster
Python version: 3.7.3
Pi model: Raspberry Pi 4 Model B Rev 1.2
GPIO Zero version: 1.6.2
Pin factory used: RPiGPIO

lurch, 2022-01-19:
ping @waveform80

EduardoNigro, 2022-01-20:
Hi,

Did you try this? I know it should be equivalent to what you did. But just in case...

```
adc1 = MCP3008(channel=0, clock_pin=11, mosi_pin=10, miso_pin=9, select_pin=8)
adc2 = MCP3008(channel=0, clock_pin=11, mosi_pin=10, miso_pin=9, select_pin=7)
```

Eduardo

slghb, 2022-01-20:
Hi Eduardo,

same error also with your code.

BrackebuschK, 2022-03-08:
I'm also experiencing the same exact issue--currently using two hardware SPI devices with two different select pins, but it causes the "pin is already in use" error after the first device reserves its pins. Here's hoping there's a fix!

slghb, 2022-03-10:
I worked around this with the following code for now:

```
def fix_gpiozero_dual_cs_spi():
    from gpiozero import Device
    from gpiozero.pins.local import (
        LocalPiHardwareSPI,
        LocalPiHardwareSPIShared,
    )
    from gpiozero.pins.rpigpio import RPiGPIOFactory

    class LocalPiHardwareSPIFixed(LocalPiHardwareSPI):
        def _conflicts_with(self, other):
            return not (
                isinstance(other, LocalPiHardwareSPI)
                and (self._port, self._device) != (other._port, other._device)
            )

    class LocalPiHardwareSPISharedFixed(
            LocalPiHardwareSPIShared, LocalPiHardwareSPIFixed
    ):
        pass

    class FixedRPiGPIOFactory(RPiGPIOFactory):
        def _get_spi_class(self, shared, hardware):
            if hardware:
                if shared:
                    return LocalPiHardwareSPISharedFixed
                return LocalPiHardwareSPIFixed
            return super()._get_spi_class(shared, hardware)

    Device.pin_factory = FixedRPiGPIOFactory()

fix_gpiozero_dual_cs_spi()
```

BrackebuschK, 2022-03-11:
Thank you @slghb ! This code is exactly what I needed in order to get things working. I really appreciate the help.

EduardoNigro, 2022-03-11:
Pretty slick! Thanks 👍

lurch, 2022-03-12:
Would you like to create a PR @slghb ?  (although I believe the maintainers of this library are busy with other projects, so I dunno when it would get merged)