gpiozero/gpiozero #1016: Software vs Hardware SPI

EduardoNigro, 2022-01-20:
Hello,

First of all GPIO Zero is very well done! 

With respect to SPI devices, it appears that the Raspberry Pi supports hardware SPI on the following pin combinations:

- SPI0: MOSI (GPIO10); MISO (GPIO9); SCLK (GPIO11); CE0 (GPIO8), CE1 (GPIO7)
- SPI1: MOSI (GPIO20); MISO (GPIO19); SCLK (GPIO21); CE0 (GPIO18); CE1 (GPIO17); CE2 (GPIO16)

However, after some testing, I noticed that GPIO Zero only treats the pins on the first bullet item (SPI0) as a hardware implementation, as per the documentation, by the way. I was wondering how difficult it would be to also have a hardware implementation for SPI1. That would really increase the number of ADC chips that could simultaneously benefit from the higher communication speed.

I would love to hear your thoughts on this subject.
Regards,
Eduardo

PS. I am writing a blog on [data acquisition](https://thingsdaq.org/) and the MCP3008 is the subject of my next post!

slghb, 2022-01-20:
Hi Eduardo,

I think the gpiozero source code at this moment only knows about the first hardware SPI interface:

https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/pi.py#L29

But maybe it's enough to just add the second interface to this dict to get it working.

EduardoNigro, 2022-01-20:
Thanks for the idea. But it didn't work :(
I got the **_failed to initialize hardware SPI, falling back to software_** message ...

I might dig into it another time.

``` python
SPI_HARDWARE_PINS = {
    0: {
        'clock':  11,
        'mosi':   10,
        'miso':   9,
        'select': (8, 7),
    },
    1: {
        'clock':  21,
        'mosi':   20,
        'miso':   19,
        'select': (18, 17, 16),
    },
}

```

lurch, 2022-01-20:
There's also even more hardware SPI interfaces on the Raspberry Pi 4B and Raspberry Pi 400 https://elinux.org/RPi_BCM2711_GPIOs

Perhaps GPIO Zero relies on the appropriate DeviceTree stuff already being configured first? :man_shrugging:  https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/overlays/README#L3124 and https://www.raspberrypi.com/documentation/computers/configuration.html#part3

slghb, 2022-01-21:
I think you have to configure the appropriate device tree settings (i.e. enable spi1 as described in the links @lurch gave) as a prerequisite for gpiozero to work with the hardware interface.

EduardoNigro, 2022-01-21:
Thanks folks! It makes sense. I'll definitely try it out when I have some time.
I'll leave this open for now until I get a chance to test it (and just in case we hear back from the developer in the meantime).

ddi-acassidy, 2022-11-03:
It would also be nice to have hardware SPI but with software chip select, nice if the bus has a lot of devices on it. Its easy to fake by just pulling a GPIO low but it would be nice to have built in.