gpiozero/gpiozero #1022: Support non-raspberry pi devices

matthijskooijman, 2022-01-24:
I'm currently using gpiozero as a convenient library on rasbperry pi, but am considering to move to an orange pi or some similar board. With lgpio support in gpiozero it should be possible to keep using the convenient API offered by gpiozero, but right now the library insists on retrieving info about the RPI device based on the revision code (i.e. I get `gpiozero.exc.PinUnknownPi: unknown old-style revision "0"`). 

It seems that it would be reasonable to use gpiozero on other boards, maybe doing without the system info and pin mapping features that are now available.

I tried this on an Orange Pi PC, and if I edit the code to just provide a fake revision, I can read GPIO pins as expected using lgpio. 

For doing this properly, I guess that the revision detection should actually check the DT or cpuinfo that it is running on a rpi (currently, I think a non-rpi with a non-zero revision will be detected as if it was some rpi based on the revision. My CPU had a revision of 0, which is invalid for rpi, so produced an error). If it is a non-rpi, the entire PiBoardInfo construction could probably be sidestepped, or one could be created with empty values?

lurch, 2022-01-24:
See also https://github.com/gpiozero/gpiozero/discussions/1018

matthijskooijman, 2022-01-24:
Ah thanks, I had searched open issues, but hadn't realized discussions are a thing.

In that discussion, though, it is suggested to make a custom Pi(n)Factory to support a specific board, but I would suggest that with the lgpio backend, *any* board should be supportable without any special support for that specific board, which is what I'm suggesting here.

Also, pin and board info discovery can technically be overridden by the PiFactory / pin backend, but I believe that currently the board info is gathered by the LocalPiFactory superclass, rather than specific subclasses. It also would not really make sense to put this info gathering in the specific subclasses, since there are multiple subclasses that work for raspberry pi boards, and even though there is currently only one that works with other boards (lgpio), making a custom e.g. OrangePiFactory a subclass of the lgpio factory does not seem correct, since in the future other factories might work with this board as well (e.g. a libgpiod-based factory is planned, I've read).

lurch, 2022-01-24:
I think there's only one person that can help you here... :point_right: @waveform80

bennuttall, 2022-01-24:
I'm closing this issue as it isn't going to get done by us. The framework is there for others to extend gpiozero for other uses, plus all the comprehensive documentation, but we can't really offer support. We barely have the capacity to add more features to the library; we certainly don't have the capacity to support other boards.

I would recommend you stick with Raspberry Pi. The main reason it's so popular is that it has a community that build things for it. There are something like 30 million Raspberry Pis in the world, and probably not many more than 200k of any other particular board.

(hope this doesn't come across rude or mean, I'm just being straight with you)

closed 2022-01-24 by bennuttall

matthijskooijman, 2022-01-25:
> (hope this doesn't come across rude or mean, I'm just being straight with you)

Not at all, I've maintained enough projects to know how this works :-)

> I would recommend you stick with Raspberry Pi. The main reason it's so popular is that it has a community that build things for it. There are something like 30 million Raspberry Pis in the world, and probably not many more than 200k of any other particular board.

I would very much love to, but they are currently hard to obtain in the numbers that this project needs, so I'm forced to look at alternatives.

> we certainly don't have the capacity to support other boards.

My suggestion would not be to support other boards specifically, but just support a mode where the current board is not known, possibly shifting some of the responsibility (e.g. for figuring out pin numbering) to the user. AFAICS this should fairly easily be doable (though maybe the detected board info is actually used for things that cannot be easily omitted, TBH I haven't checked in detail yet).

I'm also not expecting you to actually do the work here (though it *would* be convenient of course), if it seems feasible and you are open to accepting contributions in this area, I might consider making the needed changes in gpiozero myself and submitting them.

Alternatively, I could also stop using gpiozero and use lgpio directly (which would require some more code in my project, but not that much more). I'll have to see.

Thanks in any case!

dtseiler, 2023-03-25:
@matthijskooijman any interest in forking off to make your code releasable? I'm in the same boat with a couple of Libre Renegades that I had hoped to use for GPIO only to find too late that the libraries I had been using are hard-coded for the RPi.

matthijskooijman, 2023-05-29:
@dtseiler The code above is intended to be used in an application and can be used without modifications to gpiozero itself, so no forking is needed.

As for integrating the above code into gpiozero itself to support other devices, I'm not sure if this would be the right approach (it might be - I'm not just sure) and I can't spare the time to figure this out and submit a proper PR, so I'll leave that for others to pick up if they want.