gpiozero/gpiozero #1023: https://gpiozero.readthedocs.io is defaulting to the older "stable" version instead of newer "master" version

jgrimard, 2022-01-31:
Please update readthedocs.io to default to "master" instead of "stable" version.  There are changes to master that are over a year old that do not show unless the user manually selects "master".

An admin will need to log into the readthedocs.io admin page, go to the Advanced Settings link under the Admin menu of your project home, choose master in the “Default version*” dropdown, and hit Save at the bottom. Done!

lurch, 2022-01-31:
I **suspect** that the default version of the docs on ReadTheDocs matches up with the version of GPIO Zero installed on Raspberry Pi OS ?

jgrimard, 2022-01-31:
> I **suspect** that the default version of the docs on ReadTheDocs matches up with the version of GPIO Zero installed on Raspberry Pi OS ?

I don't know, but it would be nice to default to the latest version online.  
One example change that doesn't show up on ReadTheDocs "Stable":
https://github.com/gpiozero/gpiozero/pull/945

bennuttall, 2022-01-31:
No, sorry, the way it's set up is right. The master branch is not even released, and it's not expected that people will be using it or that they should rely on it. It will cause more confusion for people if we were to default to master. Anyone who jumps through the necessary hoops to install from master will be able to find their way to the appropriate docs.

closed 2022-01-31 by bennuttall