gpiozero/gpiozero #1026: Falling edge vs Power off?

Inchworm333, 2022-02-24:
I learned that my previous problem #1024 was being caused because it is checking for the falling edge and it gets triggered when the power turns off. How could I make my code tell the difference between a falling edge and the power for the switch / button being turned completely off?

lurch, 2022-02-25:
Huh? Doesn't your python code automatically stop running when the power is switched off? None of the Raspberry Pis that I have keep running after I switch them off! :rofl: 
Or perhaps you could explain your problem / setup in more detail?

Inchworm333, 2022-02-25:
I should have explained this more. The raspberry pi never turns off. We have another device that has both a switch and a button that we wanted to know the status of so we spliced the wires for them. They are both pull down and we are not able to change that. When we shut off the second device (the one with the switch and button) my code running on the raspberry pi thinks the button and switch are toggled on forever. We can get a power off command from the device but it runs after the switch and button are powered off.

lurch, 2022-02-25:
That sounds more like an electrical / wiring problem than a GPIO Zero problem? :shrug: 
Perhaps you could assume that a button-press for longer than _N_ seconds is a power-off event rather than a button-push event?

Inchworm333, 2022-02-27:
I figured out the problem. We soldered the wire to the wrong side of the switch.

closed 2022-02-27 by Inchworm333