gpiozero/gpiozero #1038: First KeyError: 17 and then AttributeError: module 'lgpio' has no attribute 'SET_BIAS_DISABLE' when executing the led blinking code after changing pinfactory to lgpio.

ShahudullahKhan, 2022-03-21:
Operating system: Raspbeery Pi OS 64 bit ("Debian GNU/Linux 11 (bullseye)")
Linux kernel version: Linux raspberrypi 5.15.28-v8+ #1532 SMP PREEMPT Tue Mar 15 13:19:15 GMT 2022 aarch64 GNU/Linux
Python version: 3.9.2
Pi model: Pi 4 Model B
GPIO Zero version: 1.6.2
Pin factory used: lgpio

I am just trying to implement led blinking code from gpiozero library after changing the pinfactory to lgpio. But after executing the code it is throwing error as in the image below.

```
```
[image]

bennuttall, 2022-03-27:
@waveform80 taking a look

andersix, 2022-05-08:
Same here. I just built "lg" and lgpio  from https://abyz.me.uk/lg/download.html, and I get the same attribute error. Note, in my code ```fac_loc = LGPIOFactory(chip=0)``` after importing ```LGPIOFactory``` from gpiozero.

```
Traceback (most recent call last):
  File "led_pwm.py", line 193, in <module>
    pb = pwm_leds()
  File "led_pwm.py", line 35, in __init__
    self.led = PWMLED(pin=LEDPIN, initial_value=1.0, frequency=PWMF, pin_factory=fac_loc)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 108, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 404, in __init__
    pin, active_high, initial_value=None, pin_factory=pin_factory
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 83, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 85, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 103, in pin
    pin = self.pin_class(self, n)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/lgpio.py", line 125, in __init__
    self.factory._handle, self.number, lgpio.SET_BIAS_DISABLE)
AttributeError: module 'lgpio' has no attribute 'SET_BIAS_DISABLE'
```

guppy, 2022-10-06:
Oops, I didn't see this issue before I did my pull request but https://github.com/gpiozero/gpiozero/pull/1063 fixes this problem.

Klusio19, 2023-11-18:
I still have the same problem. When launching python script I get:
```
path-to-my-venv/lib/python3.9/site-packages/gpiozero/devices.py:295: PinFactoryFallback: Falling back from lgpio: module 'lgpio' has no attribute 'SET_BIAS_DISABLE'
  warnings.warn(
```
but it's warning, and script does run successfully.
pip3 list:
```
Package       Version
------------- -------
colorzero     2.0
gpiozero      2.0
install       1.3.5
lgpio         0.0.0.2
pigpio        1.78
pip           23.3.1
pkg_resources 0.0.0
RPi.GPIO      0.7.1
setuptools    44.1.1
```

dhruvhaldar, 2023-12-23:
same here, the fix suggested by another user also gives the error 

.`..gpiozero/devices.py:295: PinFactoryFallback: Falling back from lgpio: module 'lgpio' has no attribute 'SET_PULL_NONE'`

dwatrous, 2023-12-23:
This started happening for me when I moved from Debian bullseye to bookworm.

lurch, 2023-12-23:
Are you using gpiozero 2.0 or an older version?
This issue was fixed in https://github.com/gpiozero/gpiozero/commit/e143923925bc1f54cc5002c73dff022f72137428

blackfist, 2024-01-03:
Same problem on my raspberry pi

```
uname -v
#1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24)

pip list
Package     Version
----------- -------
cffi        1.16.0
colorzero   2.0
gpiozero    2.0
lgpio       0.0.0.2
numpy       1.26.3
pigpio      1.78
pip         23.0.1
playsound   1.3.0
pycparser   2.21
RPi.GPIO    0.7.1
setuptools  66.1.1
sounddevice 0.4.6
soundfile   0.12.1

python
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import LED
>>> led = LED(27)
/home/blackfist/klondike/lib/python3.11/site-packages/gpiozero/devices.py:295: PinFactoryFallback: Falling back from lgpio: module 'lgpio' has no attribute 'SET_BIAS_DISABLE'
  warnings.warn(
```

lurch, 2024-01-03:
@blackfist Do you also get the same error outside of a venv, using the system-installed modules? What Pi are you using, and what OS (and version) are you running?

blackfist, 2024-01-04:
Good thinking @lurch - when I run this using the native python rather than in venv I do not get the warning. 

```
uname -a
Linux raspberrypi 6.1.0-rpi7-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.63-1+rpt1 (2023-11-24) aarch64 GNU/Linux

python
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import LED
>>> led = LED(27)
>>> 
```

ibrewster, 2024-01-04:
> Good thinking @lurch - when I run this using the native python rather than in venv I do not get the warning.

Interesting point. I am running into the same thing, also in a venv. If I do a `pip show lgpio` on the native python, I get the following:

```
Name: lgpio
Version: 0.2.2.0
Summary: Linux SBC GPIO module
Home-page: http://abyz.me.uk/lg/py_lgpio.html
Author: joan
Author-email: joan@abyz.me.uk
License: unlicense.org
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: 
```

However, if I do the same on my venv pip installed lgpio, I see this:

```
Name: lgpio
Version: 0.0.0.2
Summary: Linux SBC GPIO module
Home-page: http://abyz.me.uk/lg/py_lgpio.html
Author: joan
Author-email: joan@abyz.me.uk
License: unlicense.org
Location: /home/israel/HomeLinkBridge/venv/lib/python3.11/site-packages
Requires: 
Required-by: 
```

So the venv is version 0.0.0.2, while the system has version 0.2.2.0. Probably explains everything... now to figure out how to get the venv version up-to-date.

lurch, 2024-01-04:
Curious. https://pypi.org/project/lgpio/ says that `pip` should be able to install `lgpio 0.1.0.0`, but that page _also_ says:
```
pypy can not handle Linux libararies properly at the moment.

You need to install the lgpio module from the lg archive.

See http://abyz.me.uk/lg/download.html
```
and I've no idea how or if that would work with a venv.

bennuttall, 2024-01-04:
We were going to upload a working wheel for lgpio - @waveform80?

TechnoFeather, 2024-01-11:
I've got the same issue. Sorry, as a novice user I find it confusing that the latest version on the homepage ist `lgpio 0.1.0.0`, but on the native installation on Raspi OS 12 it's `lgpio 0.2.2.0`?

Could one ´(just for now) copy the 0.2.2.0 files over to the virtual enviroment?

majuss, 2024-02-09:
So I guess @TechnoFeather died a tragic death so he couldn't copy over the files :( RIP.
For everyone else: yes you can just copy als the lgpio und _lgpio files from /usr/lib/python/ into your venv and use version 0.2.2 that way.

GeorgeAtKistler, 2024-02-12:
@majuss I think your response is quite rude! @TechnoFeather even sais that he es a novice user and is unsure if his approach would be ok.

softlion, 2024-02-15:
So the default lgpio is now obsolete.

Should you switch to https://github.com/wiringX/wiringX ?

ezisberta, 2024-02-23:
Hi, I'm getting the same error when running a very simple light sensor project. However, the sensor seems to be working fine and after having read through some of the solutions in here and tried a few fixes myself, it feels like I could just live with this glitch? Therefore I ask, what functionality does this impact specifically, can anyone tell? Worried this might bite me back later on

chcr01, 2024-02-23:
Same happened to me using venv

lgpio 0.0.0.2

I solved it suppressing warnings by

```
import warnings
warnings.simplefilter('ignore')
import gpiozero
```

Reference: https://gpiozero.readthedocs.io/en/stable/faq.html

oxivanisher, 2024-03-09:
I now also get this error. My project worked without a problem (system packages where current before the failure) and then i updated the pi kernel. I don't know if this helps debug the issue...
But sadly no tricks from above did work as far as I can tell.

Subuday, 2024-03-15:
Yeap problem reproduced if virtual env is used.

mwshafer, 2024-03-27:
I've got the same issue when I am using a venv. I was able to get my GPIO pins to work with and LED when not using the venv, but if fails with the venv. I can confirm that my lgpio version is 0.0.0.2 in the venv and 0.2.2.0 in the native installation. What has to be done so that `pip3 install lgpio` in the venv installs the correct version? For now, I can just copy the files as mentioned by @TechnoFeather, but this isn't sustainable

ibrewster, 2024-03-27:
I just took the easy way out of allowing the venv to use the system libraries, and NOT pip installing lgpio. Then I can install everything else I need into the venv, but still use the system lgpio without copying any files.

mwshafer, 2024-03-29:
I've done the same thing that @ibrewster did. Delete the old venv and then create a new one with access to the system libraries:
`python3 -m venv venv --system-site-packages`

I'd love to know why installing lgpio in the a venv and not using the system site package installs 0.0.0.2.

lurch, 2024-04-02:
> I'd love to know why installing lgpio in the a venv and not using the system site package installs 0.0.0.2.

Looking at https://pypi.org/project/lgpio/#history , I wonder if things might be working better now? (it seems there was a new release recently)

mwshafer, 2024-04-24:
I can confirm that running `pip3 install lgpio` now installs version 0.2.2.0!