gpiozero/gpiozero #1039: Software debounce drops edge events

richvdh, 2022-04-26:
Operating system: Raspbian Stretch
Python version: 3.5 (sorry!)
Pi model: Pi Zero W Rev 1.1
GPIO Zero version: 1.4.1
Pin factory used: RPiGPIO

At least when used with RPiGPIO, software debounce for edge detection appears to drop edges, when the *previous* pulse lasted less time than the bounce time.

For example, suppose `bounce` is set to `1.0` (seconds) on an active-high pin:
 * At T=0, the pin goes high, triggering `when_activated` immediately.
 * At T=0.1, the pin goes low, but the change is (correctly) suppressed.
 * At T=5, the pin goes high again. At this point, I would *expect* `when_activated` to be called. It is not.

richvdh, 2022-04-26:
I think the problem is in `EventsMixin._fire_events`: since the falling edge is suppressed, `pin._last_active` is never cleared, so the condition at https://github.com/gpiozero/gpiozero/blob/v1.6.2/gpiozero/mixins.py#L396 ignores the next rising edge.

richvdh, 2022-04-26:
Possibly this is the same thing as #697?

lurch, 2022-04-26:
From the description in your first message, perhaps the debounce logic assumes that the "button press time" is always longer than the "debounce time"? :man_shrugging:

richvdh, 2022-04-27:
It seems like the debounce logic does assume that, yes, but it's not very useful for it to do so, and it's not what the ascii-art diagram at https://gpiozero.readthedocs.io/en/stable/api_pins.html?highlight=bounce#gpiozero.Pin.bounce says.

matthijskooijman, 2023-03-22:
Sounds lik after the 1-second bounce time, the pin should be checked and if it is still inactive, a deactivation event should be triggered to keep the state correct (and ensure matching activations with deactivations and not unbalancing the edges)?

matthijskooijman, 2023-03-22:
I've seen a similar thing with the lgpio backend, which also sometimes omits the last edge (though in that case it does not seem a debouncing issue - lgpio implements pretty much exactly what I suggested in my previous comment - but it seems that the kernel simply does not report all edges, maybe because of button bouncing combined with an ISR that is slightly too slow or so). See https://github.com/joan2937/lg/issues/18 for this issue on lgpio side.

I've looked into fixing this issue on gpiozero, and I had hoped that it would be sufficient to just read the value of the pin directly after the last edge callback, but it seems that in the case where the last rising (in my case - active low button) is omitted, directly reading the value still shows the value as high (which suggests that a single-item ISR "queue" might not be the underlying problem here). This probably means that the pin should be read "a little" after the last edge, or maybe periodically, but that's probaly more complex than I want to tackle right now (I'm mostly interested in #697, where the missing edges cause extra "when_held" calls, which probably *is* easy to fix since there is a HoldThread already that periodically triggers "when_held". I have a rough fix, I'll comment it at #697 in a minute).