gpiozero/gpiozero #1045: Motor pin problem

digitaldick, 2022-05-22:
Operating system: Ubuntu 22.04 
Python version: 3.10.4
Pi model: Pi 4 Model B Rev 1.1
GPIO Zero version: 1.6.1
Pin factory used: a003111
Works with pins 10 and 9 produces the failure below if pins 8 and 7 are selected

```python
print("from gpiozero import Motor
from time import sleep
#motor = Motor(forward=10, backward=9)
motor = Motor(forward=8, backward=7)
while True:
    motor.forward()
    sleep(5)
    motor.backward()
    sleep(5)
")
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 101, in pin
    pin = self.pins[n]
KeyError: 8

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/my_play/motor8-7.py", line 4, in <module>
    motor = Motor(forward=8, backward=7)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 108, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 1214, in __init__
    ('forward_device', PinClass(forward, pin_factory=pin_factory)),
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 108, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 403, in __init__
    super(PWMOutputDevice, self).__init__(
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 83, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 85, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 103, in pin
    pin = self.pin_class(self, n)
  File "/usr/lib/python3/dist-packages/gpiozero/pins/lgpio.py", line 124, in __init__
    lgpio.gpio_claim_input(
  File "/usr/lib/python3/dist-packages/lgpio.py", line 748, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
  File "/usr/lib/python3/dist-packages/lgpio.py", line 461, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'

digitaldick, 2022-05-22:
It's no doubt screaming at me but I'm sorry can't see it  :)

lurch, 2022-05-22:
> lgpio.error: 'GPIO busy'

Is there some other background-program which is already using pin 8?

digitaldick, 2022-05-22:
No. That was my thought. My earlier/older GPIO programs run fine.  This is the very first thing run directly after a reboot.

lurch, 2022-05-22:
I've not used Ubuntu on a Pi myself, so this is probably another one for @waveform80

digitaldick, 2022-05-22:
I don't understand that.  Is there something I should do?

lurch, 2022-05-23:
Sorry, I meant that I can't offer any further help myself, so you'll have to wait until @waveform80 is able to provide some assistance.
So no, nothing that you need to do yoursefl.

digitaldick, 2022-05-23:
A relief - thank you. In the interim I made a completely fresh system which displays the same problem.  I'm investigating the use of other pins.

joan2937, 2022-07-09:
GPIO 7 and 8 are both SPI pins.  Perhaps you have SPI enabled?  See if `ls /dev/spi*` reports any devices.

edson3x4, 2024-07-17:
i had same issue, i just had  to disable SPI from raspi-config 
Thanks