gpiozero/gpiozero #1046: Audio Connection Breaks Distance Sensor / Code Somehow

tillmanjex, 2022-06-04:
Operating system: Raspbian Bullseye
Python version: 3.9.2
Pi model: Pi 3b
GPIO Zero version: 1.6.2
Pin factory used: RPiGPIO
Sensor: HC-SR04

```python
# Using the HC-SR04 sensor

import RPi.GPIO as GPIO
from time import sleep
from gpiozero import DistanceSensor

sensor = DistanceSensor(24, 21)

while True:
    print("Distance (cm): ", sensor.distance * 100)
    sleep(0.1)
```

Hello. Strange one here... 
Case 1: Run above code _without_ 3.5mm audio cable plugged into raspi audio output. 
Result 1: Runs fine

Case 2: Run above code _with_ 3.5mm audio cable plugged into raspi audio output but _without_ being connected to audio interface
Result 2: Runs fine

Case 3: Same as case 2 but _with_ the other end of the audio cable plugged into my audio interface
Result 3: ```('no echo received')```

Case 4: Plugging in headphones instead of output cable
Result 4: Runs fine

The Audio cable is going directly into my audio interface and not passing any wires / going through an external mixer in between, etc. ie, direct uninterrupted connection. 

I highly expect this is hardware related. Maybe a ground issue? 
I would think that it wouldn't be a power / impedance issue as it worked fine with headphones, which will require more power than outputting to an audio interface channel. 

Below is the output code from Case 3:

```
sudo python3 /home/pi/WOMB-Installation/raspi-files/python/audiotest.py
/usr/lib/python3/dist-packages/gpiozero/input_devices.py:855: PWMSoftwareFallback: For more accurate readings, use the pigpio pin factory. 

See https://gpiozero.readthedocs.io/en/stable/api_input.html#distancesensor-hc-sr04 for more info
  warnings.warn(PWMSoftwareFallback(/usr/lib/python3/dist-packages/gpiozero/input_devices.py:978: DistanceSensorNoEcho: no echo received
  warnings.warn(DistanceSensorNoEcho('no echo received'))

^CTraceback (most recent call last):
  File "/home/pi/WOMB-Installation/raspi-files/python/audiotest.py", line 11, in <module>
    print("Distance (cm): ", sensor.distance * 100)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 908, in distance
    return self.value * self._max_distance
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 918, in value
    return super(DistanceSensor, self).value
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 325, in value
    return self._queue.value
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 585, in value
    self.full.wait()
  File "/usr/lib/python3.9/threading.py", line 574, in wait
    signaled = self._cond.wait(timeout)
  File "/usr/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
```

lurch, 2022-06-04:
Just to rule out any pin-weirdness, could you modify your "Non gpiozero code" to use the exact same GPIO numbers as your GpioZero examples?

> I can't successfully import pigpio library. Not sure why.

Have you installed (and started) pigpiod? What error message do you get?

tillmanjex, 2022-06-04:
You replied just as I also realised the missmatch in GPIO numbering between examples...
**BUT** (without the audio cable connection) the code still ran and read distances completely fine despite the incorrect trigger pin number...
The correct pin was 23 (ie thats what is going to Trig on the sensor).

I will have a look at the pigpio import solution later on and report back.

Thank you! 🙏

closed 2022-06-04 by tillmanjex

lurch, 2022-06-04:
> the code still ran and read distances completely fine despite the incorrect trigger pin number

Yeah, that's weird. But I'm glad that you managed to solve your original problem!