gpiozero/gpiozero #1048: servo not responding in funktions

MeiterHannes, 2022-06-18:
HI i am  new to hardware oriented programming so i am probably just making a stupid mistake but when i put a servo command inside a funktion and then request it with a button press the servo does nothing
[problemcode.txt](https://github.com/gpiozero/gpiozero/files/8933411/problemcode.txt)

ukBaz, 2022-06-18:
RPi.GPIO library is installed and because this is the default pin implementation you can use it simply by specifying an integer number for the pin in most operations.

I don't think you need the while loop as you a button press to give you max and a release to give you mid.
```
from signal import pause

from gpiozero import Servo, Button

button = Button(17)
servo = Servo(14)

servo.mid

button.when_pressed = servo.max
button.when_released = servo.mid

pause()
```
If you want to use the while loop then the second example in the docs might be an example to base it on.
https://gpiozero.readthedocs.io/en/stable/recipes.html#servo

For example:
```
from gpiozero import Servo, Button

servo = Servo(14)
btn = Button(17)

while True:
    servo.mid()
    btn.wait_for_press()
    servo.max()
    btn.wait_for_release()
```

MeiterHannes, 2022-06-18:
but is there another way to add a time delay in between the button press and the servo triggering?

bennuttall, 2022-06-18:
> RPi.GPIO library is installed and because this is the default pin implementation you can use it simply by specifying an integer number for the pin in most operations.

@ukBaz he's using pigpio which is recommended for use with servos (not RPi.GPIO) so that's ok

```python
def servo_release():
    sleep(0.5)
    servo.mid
```

@MeiterHannes The problem here is that `servo.mid` is a function/method which needs to be called with `servo.mid()`:

```python
def servo_release():
    sleep(0.5)
    servo.mid()
```

The reason the brackets are not needed here:

```python
button.when_pressed = servo.max
button.when_released = servo.mid
```

is that these are for setting callbacks - the functions are passed and called later on button events.

Also as @ukBaz said:

> I don't think you need the while loop as you a button press to give you max and a release to give you mid.

So your original code is nearly correct, just add brackets to `servo.mid` and lose the while loop:

```python
def servo_release():
    sleep(0.5)
    servo.mid()

button.when_pressed = servo.max
button.when_released = servo_release
```

Alternatively, using @ukBaz's approach, but with a delay:

```python
while True:
    servo.mid()
    btn.wait_for_press()
    sleep(0.5)
    servo.max()
    btn.wait_for_release()
    sleep(0.5)
```