gpiozero/gpiozero #1052: Be able to use a list to define the pins for ButtonBoard

Kayakbabe, 2022-07-17:
Short description of change: be able to define ButtonBoard pins with a list
Where would this change be made: ButtonBoard

Purpose of change:
Allow defining a ButtonBoard from a configuration list of values instead of having to hardcode them.

Please provide example usage of gpiozero with your suggested change: 

```python

from gpiozero import ButtonBoard

buttons_config = {
6: "next",
5: "prev",
19:"up",
26:"down",
0:"play",
16:"home",
21:"hiddenScreen",
17:"reboot"
}

#bb = ButtonBoard(19,26,4,hold_time=3,hold_repeat = False)

bb = ButtonBoard(buttons_config.keys(),hold_time=3,hold_repeat = False)

# or 
myPins = buttons_config.keys()
bb = ButtonBoard(myPins,hold_time=3,hold_repeat = False)

```

lurch, 2022-07-17:
I think `bb = ButtonBoard(*buttons_config.keys(),hold_time=3,hold_repeat = False)` should do what you want? :slightly_smiling_face: 

Alternatively, you could 'invert' your buttons_config dictionary, and do something like:
```python3
from gpiozero import ButtonBoard

buttons_config = {
   "next": 6,
   "prev": 5,
   "up": 19,
   "down": 26,
   "play": 0,
   "home": 16,
   "hiddenScreen": 21,
   "reboot": 17
}

bb = ButtonBoard(hold_time=3, hold_repeat=False, **buttons_config)
```

See https://docs.python.org/3/tutorial/controlflow.html#unpacking-argument-lists

bennuttall, 2022-07-18:
Yes, what @lurch said.

If you have a list on pin numbers, unpack it with `*` like so: `ButtonBoard(*button_pins)` which is the same as e.g. `ButtonBoard(2, 3, 4, 5)`

If you have a dict mapping names to pins (the opposite of what you have), even better because you can do `ButtonBoard(**button_pins)` which is the same as `ButtonBoard(next=6, prev=5, up=19, down=26)` and then you can access them by name like so:

```
bb.next.when_pressed = do_next
bb.prev.when_pressed = do_prev
```

This doesn't seem to be covered in the docs but the `LEDBoard` and `StatusBoard` examples are similar: https://gpiozero.readthedocs.io/en/stable/recipes_advanced.html#who-s-home-indicator