gpiozero/gpiozero #1057: lgpio.error: 'GPIO busy' | when imported from another script

WerdoxDev, 2022-07-31:
Operating system: Ubuntu Server 22.04 LTS
Python version: 3.10.4
Pi model: Pi 3 Model B
GPIO Zero version: 1.6.2
Pin factory used: RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

running a simple LED sample project works totally fine when ran individually e.g: `python3 the_script.py`
but when I need to import this script in my flask app it immediately fails with `lgpio.error: 'GPIO busy'`
I cannot put `pause()` in the script since flask also needs to run. I tried running the gpio stuff on another thread but that didn't work either

Here is my script. nothing fancy
```python
from gpiozero import LED, Button

isWriteMode = False
writeModeLed = LED(19)
toggleWriteBtn = Button(26, pull_up=False)

def activated():
    global isWriteMode
    isWriteMode = not isWriteMode
    writeModeLed.value = 1 if isWriteMode == True else 0

toggleWriteBtn.when_activated = activated
```

bennuttall, 2022-07-31:
@waveform80

WerdoxDev, 2022-08-01:
Turns out it was a problem with Flask_Reloader
when the app starts flask reloads it again thus resulting in the gpio stuff initializing twice

closed 2022-08-01 by WerdoxDev

lurch, 2022-08-01:
If you want to, you can just do `writeModeLed.value = isWriteMode` (without needing the `if`) - gpiozero is smart enough to convert the values automatically.