gpiozero/gpiozero #1058: Relay Turn off if I don't set pause()

markofranjic, 2022-09-06:
Operating system: e.g. Raspbian GNU/Linux 11 (bullseye)
Python version: e.g. 3.9.2
Pi model: e.g. Pi 4 Model B
GPIO Zero version: e.g. 1.5.1
Pin factory used: e.g. RPiGPIO

Hi Team,

First of all, I want to say that you do fantastic work! I started using **gpizero** in my project and faced some problems.
I use **Fast API** in my project, and I want to turn on the relay over the API request, but when I call the API request, the relay turn on and automatically turns off. 

```python
@app.post("/{id}/turn_on")
def turn_on(id: int, db: Session = Depends(get_db)):
    result: RelayTableModel = db.query(
        RelayTableModel).filter_by(id=id).first()

    relay =  DigitalOutputDevice(pin=result.pin, active_high=result.active_high)
    relay.on()
```

but if I set **pause()** something like this 
```python
@app.post("/{id}/turn_on")
def turn_on(id: int, db: Session = Depends(get_db)):
    result: RelayTableModel = db.query(
        RelayTableModel).filter_by(id=id).first()

    relay =  DigitalOutputDevice(pin=result.pin, active_high=result.active_high)
    relay.on()
   **pause()**
```
then everything will work, but serving the API request will be unavailable. How I can turn on relay by the API request without using the method **pause()** ?

Thank you

lurch, 2022-09-06:
This is because GpioZero "tidies up" the GPIO pins that it was using when the program finishes. See #707 and the various other issues that it links to. I think your easiest workaround will be to use RPi.GPIO directly, rather than using GPIO Zero?

bennuttall, 2022-09-06:
Yes, gpiozero is not really suitable for this until #707 is addressed.

bennuttall, 2022-09-06:
Maybe it will work if you define the relay objects outside of the route:

```python
pins = (7, 8, 9, 10, 11)  # whatever GPIO pins you are using

relays = {
    pin: DigitalOutputDevice(pin=pin, active_high=False)
    for pin in pins
}

@app.post("/{id}/turn_on")
def turn_on(id: int, db: Session = Depends(get_db)):
    relay = relays[id]
    relay.on()
```

(I'm not sure why you need a database of relays?)

Then you don't need to use `pause()` as the app running will have the same effect.