gpiozero/gpiozero #1059: Getting

flycast, 2022-09-07:
Operating system: Rasperian
Python version: e.g. 3.9.2
Pi model: Pi 4 Model B
GPIO Zero version: e.g. 1.6.2
Pin factory used: e.g. gpiozero.pins.rpigpio.RPiGPIOFactory

Having an issue using gpiozero in AWS Greengrass environment. I have a simple script that just creates a button. That is where I get the error. My script:

```python
# Copyright Amazon.com, Inc. or its affiliates. All Rights Reserved.
# SPDX-License-Identifier: MIT-0
import time
import datetime
import json
import awsiot.greengrasscoreipc
from awsiot.greengrasscoreipc.model import (
    PublishToTopicRequest,
    PublishMessage,
    JsonMessage
)
from gpiozero import Button

def say_hello():
    print("Hello!")

# Configure button 
button = Button(16)
button.when_pressed = say_hello
```

My error:

```
2022-09-07T12:42:15.407Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 101, in pin. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.407Z [WARN] (Copier) com.replaced.Publisher: stderr. pin = self.pins[n]. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.407Z [WARN] (Copier) com.replaced.Publisher: stderr. KeyError: 16. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.407Z [WARN] (Copier) com.replaced.Publisher: stderr. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.407Z [WARN] (Copier) com.replaced.Publisher: stderr. During handling of the above exception, another exception occurred:. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.407Z [WARN] (Copier) com.replaced.Publisher: stderr. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.408Z [WARN] (Copier) com.replaced.Publisher: stderr. Traceback (most recent call last):. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.408Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/greengrass/v2/packages/artifacts/com.replaced.Publisher/1.0.16/example_publisher.py", line 19, in <module>. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.408Z [WARN] (Copier) com.replaced.Publisher: stderr. button = Button(16). {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.408Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 108, in __call__. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.408Z [WARN] (Copier) com.replaced.Publisher: stderr. self = super(GPIOMeta, cls).__call__(*args, **kwargs). {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.408Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 414, in __init__. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.408Z [WARN] (Copier) com.replaced.Publisher: stderr. super(Button, self).__init__(. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.408Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 433, in __init__. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.408Z [WARN] (Copier) com.replaced.Publisher: stderr. super(HoldMixin, self).__init__(*args, **kwargs). {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.409Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 165, in __init__. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.409Z [WARN] (Copier) com.replaced.Publisher: stderr. super(DigitalInputDevice, self).__init__(. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.409Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 218, in __init__. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.409Z [WARN] (Copier) com.replaced.Publisher: stderr. super(EventsMixin, self).__init__(*args, **kwargs). {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.409Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 83, in __init__. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.409Z [WARN] (Copier) com.replaced.Publisher: stderr. super(InputDevice, self).__init__(pin, pin_factory=pin_factory). {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.409Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 549, in __init__. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.409Z [WARN] (Copier) com.replaced.Publisher: stderr. pin = self.pin_factory.pin(pin). {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.409Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 103, in pin. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.409Z [WARN] (Copier) com.replaced.Publisher: stderr. pin = self.pin_class(self, n). {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.410Z [WARN] (Copier) com.replaced.Publisher: stderr. File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 111, in __init__. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.410Z [WARN] (Copier) com.replaced.Publisher: stderr. GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull]). {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.410Z [WARN] (Copier) com.replaced.Publisher: stderr. RuntimeError: Not running on a RPi!. {scriptName=services.com.replaced.Publisher.lifecycle.Run, serviceName=com.replaced.Publisher, currentState=RUNNING}
2022-09-07T12:42:15.472Z [INFO] (Copier) com.replaced.Publisher: Run script exited. {exitCode=1, serviceName=com.replaced.Publisher, currentState=RUNNING}

```

When logged in as the user the pi was provisioned with I CAN define the button ang get meaningful results manually in python:

```
python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import Button
>>>
>>> def say_hello():
...     print("Hello!")
...
>>>
>>> # Configure button
>>> button = Button(16)
>>> button.when_pressed = say_hello
>>> Hello!
Hello!
Hello!
Hello!
Hello!
```

bennuttall, 2022-09-07:
I'm not sure what's getting run where, but your error is:

```
RuntimeError: Not running on a RPi!
```

This error comes from `RPi.GPIO` which believes you're not running the code on a Pi.

Where is the code getting run?

flycast, 2022-09-07:
Thanks for the quick response!
That may be a clue. II now realize that I may have written a Advantech dedicated OS for the Uno-220 (a raspberry Pi based IOT Gateway) rather than the Raspberry Pi OS. If that is the case then the Advantech definitely has some IO drivers that may be collioding. Please allow me to write a new OS and verify my issue.