gpiozero/gpiozero #1062: RotaryEncoder - missing transitions

flycast, 2022-09-09:
First, I really like what I see with gpiozero.
I am playing with the rotary encoder. My code is working but I notice that it is missing transitions when I spin the encoder a bit faster. My application is detecting the distance a shaft rotates in an industrial application. I will need to add "clicks" as each cycle will have multiple shaft turns. The shaft rotates at about 60 RPM at most. Using Pi 4 B for the hardware. I know at some point the encoder could turn faster then the pi can keep up.

My super simple starting place code is below. Any advice on getting the best detection speed from the software and system? Oh by the way. [In the FAQ's the number 1 question](https://gpiozero.readthedocs.io/en/stable/faq.html#how-do-i-keep-my-script-running) is about adding a `pause()` to keep the script running so callbacks get called. When I do this it is flagged as an error. That is why I put the while loop at the end.

```
from gpiozero import RotaryEncoder

def forward():
    print("----------------------->>>")

def reverse():
    print("<<<-----------")
    
rotor = RotaryEncoder(21, 20, wrap=True, max_steps=180)
rotor.steps = -180

rotor.when_rotated_clockwise = forward
rotor.when_rotated_counter_clockwise = reverse

while True:
    True

```

lurch, 2022-09-09:
> adding a pause() to keep the script running so callbacks get called. When I do this it is flagged as an error.

Did you forget to add the `from signal import pause` line near the top?

flycast, 2022-09-09:
Thanks @lurch! Fixed that issue. Now about speed....

flycast, 2022-09-09:
Wow, adding the correct pause() really helped speed as well. :)

lurch, 2022-09-10:
Excellent :grinning:

bennuttall, 2022-09-10:
Using `while True` the way you did makes the CPU cycle rather than just waiting.

Does that resolve the whole issue?