gpiozero/gpiozero #1071: No error is thrown for a DistanceSensor that is not plugged in

wederer, 2023-03-15:
Input:
```python
from gpiozero import DistanceSensor

sensor = DistanceSensor(
    5,
    6,
)

print("1")
try:
    print("sensor.distance", sensor.distance)
except:
    print("sensor.distance exception")
print("2")
```

Output:
```bash
1
/home/pi/tackyplates-backend/venv/lib/python3.9/site-packages/gpiozero/input_devices.py:978: DistanceSensorNoEcho: no echo received
  warnings.warn(DistanceSensorNoEcho('no echo received'))
```

When running this script on a RPI where a DistanceSensor is not plugged in it will hang indefinitely. I would except for gpiozero to throw an exception, that could be handled gracefully. Only a warning will be issued as can seen in the output.

Am I doing something wrong or is this intended?

Thanks in advance!

lurch, 2023-03-15:
Looking at the code, I wonder if the problem is that `DistanceSensor` is a `SmoothedInputDevice` (in order to smooth out the noisy readings) where it tries to collect a queue of values, and return the _average_ of these as the `.distance`. Perhaps it's just stuck in a loop waiting for (non-`None`) values to add to the queue? :man_shrugging:

wederer, 2023-03-17:
@lurch yeah, this makes sense and is not really intended, right? If you could point me in the right direction I would be open to create a PR. Should the Sensor just throw an error, or how would you handle it?

lurch, 2023-03-26:
I dunno - I'm afraid this is all part of the "magic @waveform80 code", which I personally don't have time to investigate :wink: