gpiozero/gpiozero #1076: KY-038 microphone device

MalcolmMielle, 2023-04-16:
Device name: Microphone
Protocol: e.g. GPIO
URL of example product: https://www.phippselectronics.com/using-the-microphone-sound-sensor-module-ky-038-with-arduino/
Further information e.g. data sheet: https://sensorkit.joy-it.net/en/sensors/ky-038

There is an example code for using this sensor with `RPi.GPIO` on [this website](https://iot-guider.com/raspberrypi/small-microphone-sound-detection-module-ky-038-raspberry-pi/)

Would be quite cool to be able to use it without having to plug in an arduino :)

MalcolmMielle, 2023-04-16:
Never the output is analog so MP3008 it is

lurch, 2023-04-18:
> URL of example product: https://www.phippselectronics.com/using-the-microphone-sound-sensor-module-ky-038-with-arduino/

I had a quick look at that page, and I don't think you'd be able to use the analog output with a Raspberry Pi, as the Pi doesn't have a built-in ADC (although _many_ ADC add-on boards and [chips](https://gpiozero.readthedocs.io/en/stable/api_spi.html#analog-to-digital-converters-adc) are available for the Pi). I think you'd probably be able to use the board's digital output with GpioZero by just creating a https://gpiozero.readthedocs.io/en/stable/api_input.html#digitalinputdevice object.

Note also that if using that board with a Pi you'd want to connect it to 3.3V rather than 5V, otherwise you risk damaging the Pi's GPIO pins.