gpiozero/gpiozero #1091: Pinout shows ??? on graphic display for Pi Zero 2 W? But /proc/cpuinfo shows correct information.

Jibun-no-Kage, 2023-10-13:
Pinout shows ??? on graphic display for Pi Zero 2 W? But /proc/cpuinfo shows correct information.

,--------------------------------.
| oooooooooooooooooooo J8 +====
| 1ooooooooooooooooooo | USB
| +====
| Pi Model ??? V1.0 |
| +----+ +====
| |D| |SoC | | USB
| |S| | | +====
| |I| +----+ |
| |C| +======
| |S| | Net
| pwr |HDMI| |I||A| +======
`-| |--------| |----|V|-------'

Revision : 902120
SoC : BCM2837
RAM : 512MB
Storage : MicroSD
USB ports : 4 (of which 0 USB3)
Ethernet ports : 1 (0Mbps max. speed)
Wi-fi : False
Bluetooth : False
Camera ports (CSI) : 1
Display ports (DSI): 1

Why does it also show WiFi not present? Symptom of the same issue?

processor : 3
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 38.40
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

Hardware : BCM2835
Revision : 902120
Serial : 000000008f249728
Model : Raspberry Pi Zero 2 W Rev 1.0

lurch, 2023-10-13:
You're probably using an old version of gpiozero, from before Zero 2 W was released. If you upgrade to the latest 2.0 version of gpiozero it should be okay? https://github.com/gpiozero/gpiozero/tags
Unfortunately there's been rather a large gap between releases.

Jibun-no-Kage, 2023-10-13:
Let's see...

root@pizero5:~/gpiozero# pip3 gpiozero
ERROR: unknown command "gpiozero"
root@pizero5:~/gpiozero# apt install python3-gpiozero
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-gpiozero is already the newest version (1.6.2-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@pizero5:~/gpiozero# pip3 install gpiozero
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: gpiozero in /usr/lib/python3/dist-packages (1.6.2)
root@pizero5:~/gpiozero# ls -l
total 56
drwxr-xr-x 5 root root 4096 Oct 13 16:07 docs
drwxr-xr-x 4 root root 4096 Oct 13 16:07 gpiozero
drwxr-xr-x 2 root root 4096 Oct 13 16:07 gpiozerocli
-rw-r--r-- 1 root root 1474 Oct 13 16:07 LICENSE.rst
-rw-r--r-- 1 root root 3380 Oct 13 16:07 Makefile
-rw-r--r-- 1 root root 8492 Oct 13 16:07 README.rst
-rw-r--r-- 1 root root 1703 Oct 13 16:07 RELEASE.rst
drwxr-xr-x 2 root root 4096 Oct 13 16:07 scripts
-rw-r--r-- 1 root root 2968 Oct 13 16:07 setup.cfg
-rw-r--r-- 1 root root   38 Oct 13 16:07 setup.py
drwxr-xr-x 3 root root 4096 Oct 13 16:07 tests
-rw-r--r-- 1 root root  265 Oct 13 16:07 tox.ini
root@pizero5:~/gpiozero#

\# python3 setup.py install
[...]
Finished processing dependencies for gpiozero==2.0

Still the same result... after upgrade to 2.0...

root@pizero5:~/gpiozero# pinout
,--------------------------------.
| oooooooooooooooooooo J8     +====
| 1ooooooooooooooooooo        | USB
|                             +====
|      Pi Model ??? V1.0         |
|      +----+                 +====
| |D|  |SoC |                 | USB
| |S|  |    |                 +====
| |I|  +----+                    |
|                   |C|     +======
|                   |S|     |   Net
| pwr        |HDMI| |I||A|  +======
`-| |--------|    |----|V|-------'

Revision           : 902120
SoC                : BCM2837
RAM                : 512MB
Storage            : MicroSD
USB ports          : 4 (of which 0 USB3)
Ethernet ports     : 1 (0Mbps max. speed)
Wi-fi              : False
Bluetooth          : False
Camera ports (CSI) : 1
Display ports (DSI): 1

J8:
   3V3  (1) (2)  5V
 GPIO2  (3) (4)  5V
 GPIO3  (5) (6)  GND
 GPIO4  (7) (8)  GPIO14
   GND  (9) (10) GPIO15
GPIO17 (11) (12) GPIO18
GPIO27 (13) (14) GND
GPIO22 (15) (16) GPIO23
   3V3 (17) (18) GPIO24
GPIO10 (19) (20) GND
 GPIO9 (21) (22) GPIO25
GPIO11 (23) (24) GPIO8
   GND (25) (26) GPIO7
 GPIO0 (27) (28) GPIO1
 GPIO5 (29) (30) GND
 GPIO6 (31) (32) GPIO12
GPIO13 (33) (34) GND
GPIO19 (35) (36) GPIO16
GPIO26 (37) (38) GPIO20
   GND (39) (40) GPIO21

For further information, please refer to https://pinout.xyz/