gpiozero/gpiozero #1095: Button hold delay + hold repeat rate

asztalosdani, 2023-10-19:
I would like to set a different hold delay and a hold repeat rate for a Button, like when you hold a key on the keyboard on PC.
Currently it is not possible because the Button waits `hold_time` first, then fires every `hold_time`.

lurch, 2023-10-20:
So I guess you're asking for e.g. a `repeat_time` parameter, which is different from `hold_time`? That'd probably need changes to [HoldMixin](https://github.com/gpiozero/gpiozero/blob/master/gpiozero/mixins.py#L409) and it might be tricky to make those changes in a backwards-compatible way.

As a "temporary hack" what happens if you directly modify the Button's `.hold_time` property from inside the `.when_held` callback? I guess you'd also need to "reset" the hold time in the `.when_released` callback too.

lurch, 2023-10-20:
>  But yes, this breaks backward-compatibility. In case this is not doable, adding a new `hold_repeat_time` parameter should do the trick. If we can agree on the interface, I can whip up a PR.

I don't think we should break backwards-compatibility (especially for such a small tweak as this), as we don't want to suddenly break all of the existing user-code that's making use of GpioZero!

I've not given it _much_ thought, but perhaps an "API" something like this works in a backwards-compatible manner?
(here I've just copied the Button's `__init__` method as an example)
```python
    def __init__(self, pin=None, *, pull_up=True, active_state=None,
                 bounce_time=None, hold_time=1, hold_repeat=False,
                 hold_repeat_time=None, pin_factory=None):
        if hold_repeat and hold_repeat_time is None:
            hold_repeat_time = hold_time
        # ...and now do everything else...
```

> > As a "temporary hack" what happens if you directly modify the Button's .hold_time property from inside the .when_held callback? I guess you'd also need to "reset" the hold time in the .when_released callback too.
> 
> Thanks, it works, it will do for the time being.

Awesome! I guess for consistency, since you're setting `hold_time` to `1` in `when_released`, you should probably explicitly pass `hold_time=1` into the `Button` constructor too?