gpiozero/gpiozero #1096: Add note on `pinout` cli that the tool is provided by gpiozero

bennuttall, 2023-10-23:
It's been noted (https://github.com/pinout-xyz/Pinout.xyz/issues/474) that people may be using the `pinout` tool without knowing it's from gpiozero, and may file bugs with @Gadgetoid because we hat tip in the direction of pinout.xyz for further information.

I would like to keep the hat tip but maybe we could add a line at the top of the output (or somewhere) to indicate the origin of the tool.

Gadgetoid, 2023-10-23:
I suppose a counterpoint to this would be to add a message to pinout.xyz about gpiozero 😆 

I appreciate the hat tip but this is definitely not the first issue where there's been confusion.

roycewilliams, 2023-10-23:
Totally makes sense in retrospect! From the end-user perspective, something like this might get close:
```
pinout is part of  the gpizero project: https://github.com/gpiozero/gpiozero
Another good Pi reference is our sibling project: https://pinout.xyz/
```
... or something like that.