gpiozero/gpiozero #1097: feature: show release date and manufacturer based on board revision

roycewilliams, 2023-10-23:
(moving from https://github.com/pinout-xyz/Pinout.xyz/issues/474)

It would be useful to extend lookup tables to map revision ID to the 'Release Date' and manufacturer (as shown in the Notes field here). This would help to determine board age and origin.

I'm not sure which lists are most definitive, but this one looks solid:

https://elinux.org/RPi_HardwareHistory#Board_Revision_History

This could be merged into the existing 'Revision' line item - such as (just for example):

`Revision           : a020d3 (Q1 2018, Sony)`

lurch, 2023-10-24:
> This would help to determine board age and origin.

Is there a reason for wanting / needing to know that, or is it just curiosity? Note that the dates you're looking at are the date the model was first introduced; which will probably be different to the age of any particular board that you own, as most models of Raspberry Pi are still being produced.

> I'm not sure which lists are most definitive

This one is definitely the "most definitive" (or at least the most authoritative), but it doesn't have the date information that you're looking for https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-revision-codes

bennuttall, 2023-10-24:
Yeah this feels like bloat. The pinout tool is supposed to be just a pin label. Dave went overboard as usual and added a load of other stuff :)

closed 2023-10-24 by roycewilliams