gpiozero/gpiozero #1105: Hi， I got an error when I use this repo in Raspberry Pi 5.

FLamefiREz, 2024-01-03:
Device name: e.g. Temperature sensor
Protocol: e.g. GPIO/SPI/I2C
URL of example product:
Further information e.g. data sheet:

Please provide example code for using this device with Python, using another library:

```python
from gpiozero import DistanceSensor
from time import sleep

sensor = DistanceSensor(echo=18, trigger=17)
while True:
    print('Distance: ', sensor.distance * 100)
    sleep(1)
```

RuntimeError: Cannot determine SOC peripheral base address

closed 2024-01-03 by FLamefiREz

Ed1ks, 2024-01-13:
How did you solve that?

FLamefiREz, 2024-01-16:
> How did you solve that?

During the process of exploring Raspberry Pi 5, I mistakenly updated the built-in dependencies of Raspberry Pi OS, resulting in incompatibility with this library. This issue was identified in the official documentation. Since I was unable to resolve these incompatible dependency issues, I chose to reinstall the system, and the problem was subsequently resolved.

Ed1ks, 2024-01-16:
Thank you for your answer.
I am trying to connect to the gpio from a docker container. And what I think is  that its not possible to install dependencies, which work with rpi 5. Your Post ist underlining it, as you updated libraries, which result in incompatibility.
I am wondering, why the Rasperry Pi Os library, which is working (probably RPI.gpio) is not the official package.

lurch, 2024-02-24:
> I am wondering, why the Rasperry Pi Os library, which is working (probably RPI.gpio) is not the official package.

RPi.GPIO doesn't work on the Raspberry Pi 5.