gpiozero/gpiozero #1113: 3rd party import-metadata not supported?

GoremanX, 2024-01-25:
Operating system: Raspbian Buster (Volumio OS)
Python version: 3.7
Pi model:  Pi 5
GPIO Zero version: 2.0
Pin factory used: RPiGPIO

In a python3 venv, I used pip3 to install gpiozero and import-metadata. However when trying to run a very simple button press detection script, I get the following error:

```
Traceback (most recent call last):
  File "button-check.py", line 1, in <module>
    from gpiozero import Button
  File "/home/volumio/deadman-env/lib/python3.7/site-packages/gpiozero/__init__.py", line 19, in <module>
    from .pins import (
  File "/home/volumio/deadman-env/lib/python3.7/site-packages/gpiozero/pins/__init__.py", line 21, in <module>
    from ..devices import Device
  File "/home/volumio/deadman-env/lib/python3.7/site-packages/gpiozero/devices.py", line 19, in <module>
    from importlib.metadata import entry_points
ModuleNotFoundError: No module named 'importlib.metadata'
```
My understanding based on the documentation is that Python 3.5 and newer is supported by gpiozero 2.0, but this error seems to imply that importing libraries is being accomplished with Python 3.8's built-in import-metadata rather than the 3rd part external one. And since I'm using Python 3.7, that wouldn't work. So I guess either something wrong is happening or the documentation is misleading?

MusicalNinjaRandInt, 2024-01-30:
Hi folks, based on observations from the code it appears that python3.9 is the lowest version currently supported. A documentation adjustment (to `compat.rst`) is included in #1116 and #1112 if that's the way you wish to resolve it.

bennuttall, 2024-01-30:
This can be fixed

MusicalNinjaRandInt, 2024-01-30:
@waveform80 - I'd be interested in looking at this and overall backwards compatibility. If you're OK with that: do you want to keep the current statement that 3.5 and newer is supported or do you expect most users to be on something later, considering that 3.8 goes EOL this year, pip doesn't support 3.5 anymore, but users may well still be on older versions ...

Here's a quick initial idea of the possible scope of changes involved: https://github.com/MusicalNinjaRandInt/pytest_gpiozero/actions/runs/7710548419/job/21014085625

bennuttall, 2024-01-30:
Generally speaking we'd aim to support the Python versions in Debian, until those Debian versions go EOL, so for now that includes 3.7.

MusicalNinjaRandInt, 2024-01-30:
Gotcha. Thanks, are you OK with me looking at this and making a suggestion?

bennuttall, 2024-01-31:
Sure

MusicalNinjaRandInt, 2024-02-01:
@waveform80 , @bennuttall - I have a working branch which is passing all tests (except real_pins) for 3.7-3.12 at https://github.com/MusicalNinjaRandInt/pytest_gpiozero/tree/backwards_compatibility_working

Any comments on the approach and ideas for improvement are welcome.

I still need to look through the various pins implementations which have low coverage in the pipeline to check for any hidden imports and tidy up the docs, squash etc. before raising a PR.

softlion, 2024-02-14:
> 3.12

You used https://github.com/ewels/rich-click/issues/140 ?

Otherwise issue is at line 307 in https://github.com/gpiozero/gpiozero/blob/master/gpiozero/devices.py

waveform80, 2024-02-15:
As I've mentioned in #1118, we don't want to *officially* support anything below 3.9 (which is Debian/RaspiOS oldstable at the time of writing), and I've just pushed some commits which limit our CI config to 3.9 onwards (as it should've been already).

That said, right now there's nothing terribly complicated about including Python 3.7 support; it's basically an import of a backported lib and that's it. As a consequence, I'm not above pushing a commit that keeps us compatible with 3.7 *for now* (which I'm just about to do). However, I'm deliberately not extending our CI config to include this: we don't *officially* support 3.7, and we're not going to test for breakage on it. Please make plans to move to a supported version.

P.S. You're not using RPiGPIO, or Raspbian Buster, on a Pi 5 :)

closed 2024-02-15 by waveform80

GoremanX, 2024-02-15:
Believe it or not, I am indeed running RpiGPIO and Buster on a Pi 5. Not by choice, but that's what Volumio is based on. They only backported a Pi 5-compatible kennel into their supremely old distribution

waveform80, 2024-02-15:
> Believe it or not, I am indeed running RpiGPIO and Buster on a Pi 5. Not by choice, but that's what Volumio is based on. They only backported a Pi 5-compatible kernel into their supremely old distribution

Let me re-phrase ... you may well have an armhf buster userland (with a vastly more recent arm64 kernel jammed underneath it), and you may well have RPi.GPIO installed on that userland. But it's not going to do anything because RPi.GPIO won't talk to the RP1 controlling the GPIOs on the Pi 5 :)

commit ebf8874b (2024-02-15): Fix #1113

Unofficial 3.7 compat; no, we don't *officially* support 3.7 but so long
as it's trivial, what the hell. That said, if you're still relying on
3.7, you're on borrowed time!