gpiozero/gpiozero #1114: Basic recipe won't work with i2c servo HAT in use.

Kilted-Weaver, 2024-01-26:
When I tried to use the button basic recipe on pin 2 while using the servo HAT i2c every time i pressed the button the servos stopped responding. Moving to pin 26 solved the problem (after a while of head scratching as a newbie to python and robotics) as I noticed in the pin lay out that pin 2 has a 12c designation.

Kilted-Weaver, 2024-01-26:
Please could a note be added to warn other newbies. Thank you for all the documentation without which I would be still scratching my head.

bennuttall, 2024-01-26:
Sorry, that's a bit specific. It's generally the case that you need to exercise caution when using GPIOs alongside a HAT.

lurch, 2024-01-28:
https://pinout.xyz/boards/ Is a useful way to figure out what pins may be in use by a particular HAT. If a pin is being used by your HAT, then you shouldn't try using that pin with GpioZero at the same time.