gpiozero/gpiozero #1130: gpiozero failure run from systemd after update

jnob1, 2024-03-20:
I have an unusual problem. I have a pi4 with several GPIO connected weather-related devices. I was not having any issues before a recent apt update (which included many updates including a new kernel, but not any python modules) and rebooting, at which point my program began to crash (errors below.) The program is controlled by systemd (started at boot, restarted if it crashes, and watchdogged using sdnotify.) However, if I run the program manually (disable for systemd and run from the shell) it runs without issue. The program is run non-root, and also in the systemd config I specify the non-root user. 
Not sure where to look or how to debug this.

python3-gpiozero/stable 1.6.2-1+b1 armhf
Linux pi-station 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

Here's the log:
Mar 19 18:02:48 pi-station python3[2048]: 2024-03-19 18:02:48 xCreatePipe: Can't set permissions (436) for //.lgd-nfy0, No such file or directory
Mar 19 18:02:48 pi-station python3[2048]: /usr/lib/python3/dist-packages/gpiozero/devices.py:295: PinFactoryFallback: Falling back from lgpio: [Errno 2] No such file or directory: '.lgd-nfy-3'
Mar 19 18:02:48 pi-station python3[2048]:   warnings.warn(
Mar 19 18:02:49 pi-station python3[2048]: Traceback (most recent call last):
Mar 19 18:02:49 pi-station python3[2048]:   File "/home/myth/weather-station/weather_station_BYO.py", line 51, in <module>
Mar 19 18:02:49 pi-station python3[2048]:     rain_sensor = Button(6)
Mar 19 18:02:49 pi-station python3[2048]:                   ^^^^^^^^^
Mar 19 18:02:49 pi-station python3[2048]:   File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 103, in __call__
Mar 19 18:02:49 pi-station python3[2048]:     self = super().__call__(*args, **kwargs)
Mar 19 18:02:49 pi-station python3[2048]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 19 18:02:49 pi-station python3[2048]:   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 412, in __init__
Mar 19 18:02:49 pi-station python3[2048]:     super().__init__(
Mar 19 18:02:49 pi-station python3[2048]:   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 417, in __init__
Mar 19 18:02:49 pi-station python3[2048]:     super().__init__(*args, **kwargs)
Mar 19 18:02:49 pi-station python3[2048]:   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 168, in __init__
Mar 19 18:02:49 pi-station python3[2048]:     self.pin.when_changed = self._pin_changed
Mar 19 18:02:49 pi-station python3[2048]:     ^^^^^^^^^^^^^^^^^^^^^
Mar 19 18:02:49 pi-station python3[2048]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 471, in <lambda>
Mar 19 18:02:49 pi-station python3[2048]:     lambda self, value: self._set_when_changed(value),
Mar 19 18:02:49 pi-station python3[2048]:                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Mar 19 18:02:49 pi-station python3[2048]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 636, in _set_when_changed
Mar 19 18:02:49 pi-station python3[2048]:     self._enable_event_detect()
Mar 19 18:02:49 pi-station python3[2048]:   File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 220, in _enable_event_detect
Mar 19 18:02:49 pi-station python3[2048]:     GPIO.add_event_detect(
Mar 19 18:02:49 pi-station python3[2048]: RuntimeError: Failed to add edge detection

jnob1, 2024-03-21:
I solved the issue by adding WorkingDirectory=~ to my process's systemd service description file. I thought that was set automatically if the process was run as a non-root User, but apparently not, and why it worked before the update I don't know.

danny-andersen, 2024-03-21:
I get the same problem but running it on the command line. It appears to be an issue with running it on a Linux 6.6 kernel, which was released on 15th March for Raspbian Bookworm.