gpiozero/gpiozero #1135: LightSensor() no longer works under Raspbian Bookworm / Linux 6.6.20

danny-andersen, 2024-03-28:
Operating system:  Raspbian Bookworm
Python version: 3.11.2
Pi model: e.g. Pi 4 Model B
GPIO Zero version: 2.0
Pin factory used: lgpio version 0.2.2.0

Recent upgrade of Bookworm that uplifted the Linux vernal from 6.1.0 to 6.6.20, the LightSensor() class no longer works. Simple test programme below, the when_light() callback is never called and the LightSensor value always returns 0.0, regardless of the light level.

```python
from gpiozero import LightSensor
from time import sleep

ldr = LightSensor(17, charge_time_limit=0.04)
ldr.when_light = lambda : print("Light")
while True:
	print(f"LDR: {ldr.value}")
	sleep(1)

```
Same circuit was used before and after. Also proved that the circuit still works by testing using OutputDevice() and InputDevice() to discharge the capacitor and timing how long it takes to charge based on whether it is light or not.

bennuttall, 2024-04-01:
@waveform80?

nkreipke, 2024-05-24:
Can confirm this for both Pi Zero 2W and Pi 4. Not sure about the pull down thing though, the capacitor voltage peaks at around 3.3V as expected (measured at the GPIO pin).

[image]

Previously, the capacitor immediately got discharged when the positive edge was detected. After upgrading, the library now always waits until the `charge_time_limit` has run out, failing to detect the edge.

I've also had this issue on ArchLinux ARM. As RPi.GPIO still works there, I was able to fix it by **uninstalling** lgpio and letting gpiozero fall back to RPi.GPIO. The issue seems to be specific to lgpio.

Unfortunately, RPi.GPIO is unusable on Bookworm because of [this bug](https://github.com/raspberrypi/linux/issues/6037), so there seems to be no way to actually get it to work, which is kind of disappointing.

ZebrasHaveBigEars, 2024-08-19:
Got to agree. Does not work. I have tried on Pi 5 and the output is always 0.0

Jmi91, 2025-01-12:
Hi,
is there a solution now ?

creamailer-ape, 2025-02-07:
Is there a solution or workaround available?

lars-derichter, 2025-03-02:
Same problem. Wasted two hours in a class full of students trying to fix this.

danny-andersen, 2025-03-02:
Unfortunately, you need to do the detection manually:

from gpiozero import LightSensor, OutputDevice, Button, InputDevice

    ctx.pir = OutputDevice(pin=ctx.PIR_IN, 
            active_high=True, initial_value=False)
    # Drive output low to discharge the capacitor
    sleep(0.1)
    # Set to input device
    ctx.pir.close()
    ctx.pir = InputDevice(pin=ctx.PIR_IN, pull_up=None, 
             active_state=True)
    # Count how long it takes for the input pin to 
             recharge
    # For dark - about 0.7 seconds, for light, about 0.04s
    for i in range(1, 10):
        if ctx.pir.is_active:
            ctx.lastPirTime = nowTime.timestamp()
            if ctx.DEBUG:
                print(f"{nowTime}: PIR Triggered at count {i}")
            break
        sleep(0.01)
    ctx.pir.close()

derplarsen, 2025-04-18:
@danny-andersen I tried to run your script and got 

> NameError: name 'ctx' is not defined

Sorry, I'm pretty new to this but definitely having issues while knowing my pinouts and components are fine, even tried a different `charge_time_limit` (like above, 0.04). 

I'd like to try to run your manual detection but what is `ctx`? Thanks

danny-andersen, 2025-04-19:
Apologies, it was just a class used to hold local state. You can just delete the ctx bit, e.g. instead of ctx.pir use pir.

peppe8o, 2025-06-27:
If this can help, I fixed the issue with a Raspberry PI 4 model B by expliciting the PiGPIOFactory as pin factory.  Here's the full tutorial:
[https://peppe8o.com/using-photoresistor-from-raspberry-pi-to-detect-light/](url)

morphias2004, 2026-03-08:
Still broken. :(