gpiozero/gpiozero #1137: NativeFactory `wait_for_press` and `when_activated` callbacks not called

NeonDaniel, 2024-03-30:
Operating system: Debian Bookworm ([neon-debos](https://github.com/NeonGeckoCom/neon_debos) build)
Python version: 3.10.13
Pi model: Pi 4 Model B
GPIO Zero version: 2.0.1
Pin factory used: Native

I had a working class implemented with the default (RPi.GPIO) backend working but I am trying to move to `NativeFactory` to support kernel 6.6 and Pi5. The relevant code is linked to https://github.com/NeonGeckoCom/neon-phal-plugin-switches/pull/16.

I tried debugging interactively and noted that a `when_activated` callback was never called and `wait_for_press` would wait infinitely when I tried pressing and holding the button multiple times. `is_pressed` and `value` appear to always report accurately though.
```shell
Python 3.10.13 (main, Mar 29 2024, 21:16:14) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpiozero
>>> from gpiozero.pins.native import NativeFactory
>>> act = gpiozero.Button(24, pull_up=None, active_state=False, pin_factory=NativeFactory())
>>> # button not pressed
>>> print(act.is_pressed)
False
>>> # button pressed
>>> print(act.is_pressed)
True
>>> act.wait_for_press()
^CTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/neon/venv/lib/python3.10/site-packages/gpiozero/mixins.py", line 278, in wait_for_active
    return self._active_event.wait(timeout)
  File "/usr/local/lib/python3.10/threading.py", line 607, in wait
    signaled = self._cond.wait(timeout)
  File "/usr/local/lib/python3.10/threading.py", line 320, in wait
    waiter.acquire()
KeyboardInterrupt
>>> 
```

lurch, 2024-04-03:
> I am trying to move to `NativeFactory` to support kernel 6.6 and Pi5.

You might have better luck using `LGPIOFactory` ? See https://gpiozero.readthedocs.io/en/latest/api_pins.html#changing-the-pin-factory

NeonDaniel, 2024-04-03:
> > I am trying to move to `NativeFactory` to support kernel 6.6 and Pi5.
> 
> You might have better luck using `LGPIOFactory` ? See https://gpiozero.readthedocs.io/en/latest/api_pins.html#changing-the-pin-factory

I don't think I tried that yet but if it supports kernel 6.6+ then it could be a valid workaround.